UshakovVasilii/gnome-shell-extension-freon #2: run sensors-detect as root

Bavarianph, 2015-02-13:
When one clicks on the freon icon it gives  an error and then
notifies: Please run sensors-detect as root.
However sensors-detect is in root.
???

UshakovVasilii, 2015-02-13:
Hi, I don't understand what you mean.

    [ushakov@ushakov-archlinux ~]$ sensors-detect 
    You need to be root to run this script.

https://wiki.archlinux.org/index.php/lm_sensors

UshakovVasilii, 2015-02-13:
Ah, do you mean that you already runner `sensors-detect` under root?
If so then please provide `sensors` output

Bavarianph, 2015-02-13:
sensors-detect is root,

however, clicking on the the icon or even before the word next to freon, it
says: "Error";

clicking on "Error" the notification says: "Please run sensors-detect as
root."

And every once in a while it shows temperatures.

These temperatures change to higher and lower counts every time one points
a mouse cursor on it.

The temperatures change drastically, and then the "Error" notice shows up
again, with "Please run sensors-detect as root."

i do not know where a Freon data or log may be found on Ubuntu-Gnome 14.10.

My lighted Thermometer box built into the PC case, gives me around 26
Celsius to 30 Celsius as the temperature inside the case.

My PC is a msi-A55M-E33-AMD FM+ motherbord with quad cpus & Radeon HD 7560D.

It would be nice to have trusted numbers of my PC machine.

That is really all I can tell you.

Sincerely,

BavarianPH@gmail.com

UshakovVasilii, 2015-02-13:
Most of the data is taken from `sensors` output, so I think you have problem with `lm_sensors`.
Please check `sensors` output.

Bavarianph, 2015-02-15:
Thank you for trying to fix freon.

But it does not work for my PC setup.

BavarianPH

closed 2015-02-18 by UshakovVasilii

traptoo, 2024-07-05:
I hope this is the right place for my issue with equal error message:

After some updates of my OpenSuse Tumbleweed Laptop Freon shows up a warning Icon:
[image]
Message after clicking says "Please run sensors-detect as root.

Sensors output:
```
dell_ddv-virtual-0
Adapter: Virtual device
CPU Fan:        0 RPM
Video Fan:      0 RPM
CPU:          +52.0°C  (low  = -127.0°C, high = +95.0°C)
SODIMM:       +41.0°C  (low  = -127.0°C, high = +64.0°C)
HDD:          +42.0°C  (low  = -127.0°C, high = +66.0°C)
Ambient:      +39.0°C  (low  = -127.0°C, high = +42.0°C)
Other:        +48.0°C  (low  = -127.0°C, high = +66.0°C)
Ambient:      +42.0°C  (low  = -127.0°C, high = +57.0°C)
Ambient:      +39.0°C  (low  = -127.0°C, high = +45.0°C)

pch_cannonlake-virtual-0
Adapter: Virtual device
temp1:        +50.0°C  

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +50.0°C  

ucsi_source_psy_USBC000:003-isa-0000
Adapter: ISA adapter
in0:           5.00 V  (min =  +5.00 V, max =  +5.00 V)
curr1:         0.00 A  (max =  +0.00 A)

ucsi_source_psy_USBC000:001-isa-0000
Adapter: ISA adapter
in0:           5.00 V  (min =  +5.00 V, max =  +5.00 V)
curr1:         0.00 A  (max =  +0.00 A)

BAT0-acpi-0
Adapter: ACPI interface
in0:           7.80 V  
curr1:         2.14 A  

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +51.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +50.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +49.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +48.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +50.0°C  (high = +100.0°C, crit = +100.0°C)

dell_smm-isa-00de
Adapter: ISA adapter
fan1:           0 RPM  (min =    0 RPM, max = 7300 RPM)
fan2:           0 RPM  (min =    0 RPM, max = 7300 RPM)
temp1:        +52.0°C  
temp2:        +41.0°C  
temp3:        +42.0°C  
temp4:        +39.0°C  
temp5:        +48.0°C  
temp6:        +42.0°C  
temp7:        +39.0°C  
pwm1:              0%
pwm2:              0%

ucsi_source_psy_USBC000:002-isa-0000
Adapter: ISA adapter
in0:          20.00 V  (min =  +5.00 V, max =  +0.40 V)
curr1:         2.25 A  (max =  +3.56 A)

nvme-pci-7100
Adapter: PCI adapter
Composite:    +44.9°C  (low  = -273.1°C, high = +81.8°C)
                       (crit = +84.8°C)
Sensor 1:     +44.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +42.9°C  (low  = -273.1°C, high = +65261.8°C)
```

Freon worked just fine before updating the system (~1100 updates including kernel).
What can i do?