UshakovVasilii/gnome-shell-extension-freon #5: Support for nvidia optimus?

spacepluk, 2015-04-06:
It would be cool to show the temperature of the gpu when it's enabled.

UshakovVasilii, 2015-04-06:
Oh, I hate "optimus"...
As I understand nouveau and [nvidia without dynamic switching GPU](https://wiki.archlinux.org/index.php/NVIDIA_Optimus#Using_nvidia)  should work.
Do you use [Bumblebee](https://wiki.archlinux.org/index.php/Bumblebee)?

spacepluk, 2015-04-06:
Hehehe, yeah I used to hate it too.
I'm using bumblebee with nvidia drivers, but gnome is running on the intel. I only run certain applications with optirun when I want the nvidia to kick in.  Should it work in this situation?

For example, if I run:

    optirun glxgears

and then check the temperatures I can't see the values for the gpu.

Maybe I'm missing something.

UshakovVasilii, 2015-04-06:
Got it,
I will check with bumblebee on this week.
Thanks

spacepluk, 2015-04-06:
Thank you :)
Let me know if I you need more info/testing.

UshakovVasilii, 2015-04-11:
Done! "bumblebee-nvidia" branch. 
I added "Bumblebee + nvidia" utility (option in config).
I'm using `optirun nvidia-smi -L` for detect GPU name and `optirun nvidia-smi -q -d TEMPERATURE` for detect temperature. Commands are executed if GPU is working only.
I'll test more, merge into master and upload to extensions.gnome.org on next week.

spacepluk, 2015-04-12:
woot! thanks :)

UshakovVasilii, 2015-04-14:
uploaded to extensions.gnome.org

closed 2015-04-14 by UshakovVasilii