UshakovVasilii/gnome-shell-extension-freon #7: Extension writes to syslog approximately every 5 seconds

nroach44, 2015-04-20:
```
Apr 21 00:30:17 poseidon gnome-session[1666]: Gjs-Message: JS LOG: [FREON] Render all MenuItems
Apr 21 00:30:22 poseidon gnome-session[1666]: Gjs-Message: JS LOG: [FREON] Render all MenuItems
Apr 21 00:30:27 poseidon gnome-session[1666]: Gjs-Message: JS LOG: [FREON] Render all MenuItems
```

This has happened on the extenstion for a few versions now, but it is definitely occuring on the current version.

Debian Jessie x64, with Gnome 3.14.

UshakovVasilii, 2015-04-20:
Hi, something wrong. Extension add and remove all menu-items for each  timer tick, it shouldn't happen.
I committed more logs. Please provide output.
Thank you!

nroach44, 2015-04-21:
Updating to the current version on github has stopped the behavior. I'm not sure if it's that commit or if you fixed it earlier but haven't uploaded it to extensions.gnome.org

UshakovVasilii, 2015-04-21:
extensions.gnome.org has latest version, It is strange...

closed 2015-04-21 by UshakovVasilii