UshakovVasilii/gnome-shell-extension-freon #8: log

ersaloz, 2015-04-30:
Please remove/comment line
  global.log('[FREON] Render all MenuItems');
from extension.js, as it logs to syslog every poll interval.

UshakovVasilii, 2015-04-30:
Hi, it shouldn't write to syslog every poll interval. Something wrong...
Please install extension from `additional-logs` branch and provide logs.
Thanks!

ersaloz, 2015-04-30:
> Hi, it shouldn't write to syslog every poll interval. Something wrong...
> Please install extension from `additional-logs` branch and provide logs.

Here they are. One collects some logs from freon as installed from
extensions.gnome.org, the reason of my first post. The other one shows
the logs resulting from `additional-logs` branch.

> Thanks!
> 
> ---
> https://github.com/UshakovVasilii/gnome-shell-extension-freon/issues/8#issuecomment-97871871

Apr 30 23:03:00 escampalaboira gnome-session[4550]: (gnome-shell:4708): Gtk-WARNING **: Symbolic icon /usr/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/icons/freon-gpu-temperature-symbolic.svg is not in an icon theme directory
Apr 30 23:03:00 escampalaboira gnome-session[4550]: Gjs-Message: JS LOG: [FREON] Render all MenuItems. Last sensor count is undefined, current is 18 Sensors:
Apr 30 23:03:00 escampalaboira gnome-session[4550]: CPUTIN
Apr 30 23:03:00 escampalaboira gnome-session[4550]: SYSTIN
Apr 30 23:03:00 escampalaboira gnome-session[4550]: Average
Apr 30 23:03:00 escampalaboira gnome-session[4550]: Maximum
Apr 30 23:03:00 escampalaboira gnome-session[4550]: temperature-group
Apr 30 23:03:00 escampalaboira gnome-session[4550]: fan2
Apr 30 23:03:00 escampalaboira gnome-session[4550]: +3.3V
Apr 30 23:03:00 escampalaboira gnome-session[4550]: 3VSB
Apr 30 23:03:00 escampalaboira gnome-session[4550]: AVCC
Apr 30 23:03:00 escampalaboira gnome-session[4550]: cpu0_vid
Apr 30 23:03:00 escampalaboira gnome-session[4550]: in1
Apr 30 23:03:00 escampalaboira gnome-session[4550]: in4
Apr 30 23:03:00 escampalaboira gnome-session[4550]: in5
Apr 30 23:03:00 escampalaboira gnome-session[4550]: Vbat
Apr 30 23:03:00 escampalaboira gnome-session[4550]: Vcore
Apr 30 23:03:00 escampalaboira gnome-session[4550]: Gjs-Message: JS LOG: Added: 1
Apr 30 23:03:00 escampalaboira gnome-session[4550]: Gjs-Message: JS LOG: Added: 3
Apr 30 23:03:00 escampalaboira gnome-session[4550]: Gjs-Message: JS LOG: Added: 4
Apr 30 23:03:00 escampalaboira gnome-session[4550]: Gjs-Message: JS LOG: Activated: 4
Apr 30 23:03:00 escampalaboira gnome-session[4550]: Gjs-Message: JS LOG: [FREON] Render all MenuItems. Last sensor count is 18, current is 20 Sensors:
Apr 30 23:03:00 escampalaboira gnome-session[4550]: CPUTIN
Apr 30 23:03:00 escampalaboira gnome-session[4550]: SYSTIN
Apr 30 23:03:00 escampalaboira gnome-session[4550]: ST3500320AS
Apr 30 23:03:00 escampalaboira gnome-session[4550]: ST3500320AS
Apr 30 23:03:00 escampalaboira gnome-session[4550]: Average
Apr 30 23:03:00 escampalaboira gnome-session[4550]: Maximum
Apr 30 23:03:00 escampalaboira gnome-session[4550]: temperature-group
Apr 30 23:03:00 escampalaboira gnome-session[4550]: fan2
Apr 30 23:03:00 escampalaboira gnome-session[4550]: +3.3V
Apr 30 23:03:00 escampalaboira gnome-session[4550]: 3VSB
Apr 30 23:03:00 escampalaboira gnome-session[4550]: AVCC
Apr 30 23:03:00 escampalaboira gnome-session[4550]: cpu0_vid
Apr 30 23:03:00 escampalaboira gnome-session[4550]: in1
Apr 30 23:03:00 escampalaboira gnome-session[4550]: in4
Apr 30 23:03:00 escampalaboira gnome-session[4550]: in5
Apr 30 23:03:00 escampalaboira gnome-session[4550]: Vbat
Apr 30 23:03:00 escampalaboira gnome-session[4550]: Vcore
Apr 30 23:03:04 escampalaboira gnome-session[4550]: Gjs-Message: JS LOG: [FREON] Render all MenuItems. Cannot find [temperature-group] menu item Sensors:
Apr 30 23:03:04 escampalaboira gnome-session[4550]: CPUTIN
Apr 30 23:03:04 escampalaboira gnome-session[4550]: SYSTIN
Apr 30 23:03:04 escampalaboira gnome-session[4550]: ST3500320AS
Apr 30 23:03:04 escampalaboira gnome-session[4550]: ST3500320AS
Apr 30 23:03:04 escampalaboira gnome-session[4550]: Average
Apr 30 23:03:04 escampalaboira gnome-session[4550]: Maximum
Apr 30 23:03:04 escampalaboira gnome-session[4550]: temperature-group
Apr 30 23:03:04 escampalaboira gnome-session[4550]: fan2
Apr 30 23:03:04 escampalaboira gnome-session[4550]: +3.3V
Apr 30 23:03:04 escampalaboira gnome-session[4550]: 3VSB
Apr 30 23:03:04 escampalaboira gnome-session[4550]: AVCC
Apr 30 23:03:04 escampalaboira gnome-session[4550]: cpu0_vid
Apr 30 23:03:04 escampalaboira gnome-session[4550]: in1
Apr 30 23:03:04 escampalaboira gnome-session[4550]: in4
Apr 30 23:03:04 escampalaboira gnome-session[4550]: in5
Apr 30 23:03:04 escampalaboira gnome-session[4550]: Vbat
Apr 30 23:03:04 escampalaboira gnome-session[4550]: Vcore
Apr 30 23:03:09 escampalaboira gnome-session[4550]: Gjs-Message: JS LOG: [FREON] Render all MenuItems. Cannot find [temperature-group] menu item Sensors:
[... 282 lines cut ...]
Apr 30 22:36:36 escampalaboira gnome-session[2796]: Gjs-Message: JS LOG: [FREON] Render all MenuItems
Apr 30 22:36:41 escampalaboira gnome-session[2796]: Gjs-Message: JS LOG: [FREON] Render all MenuItems
Apr 30 22:36:46 escampalaboira gnome-session[2796]: Gjs-Message: JS LOG: [FREON] Render all MenuItems
Apr 30 22:36:51 escampalaboira gnome-session[2796]: Gjs-Message: JS LOG: [FREON] Render all MenuItems
Apr 30 22:36:56 escampalaboira gnome-session[2796]: Gjs-Message: JS LOG: [FREON] Render all MenuItems
Apr 30 22:37:01 escampalaboira gnome-session[2796]: Gjs-Message: JS LOG: [FREON] Render all MenuItems
Apr 30 22:37:06 escampalaboira gnome-session[2796]: Gjs-Message: JS LOG: [FREON] Render all MenuItems
Apr 30 22:37:11 escampalaboira gnome-session[2796]: Gjs-Message: JS LOG: [FREON] Render all MenuItems
Apr 30 22:37:16 escampalaboira gnome-session[2796]: Gjs-Message: JS LOG: [FREON] Render all MenuItems
[... cut ...]
Apr 30 22:38:16 escampalaboira gnome-session[2796]: Gjs-Message: JS LOG: [FREON] Render all MenuItems
Apr 30 22:38:21 escampalaboira gnome-session[2796]: Gjs-Message: JS LOG: [FREON] Render all MenuItems
Apr 30 22:38:26 escampalaboira gnome-session[2796]: Gjs-Message: JS LOG: [FREON] Render all MenuItems
Apr 30 22:38:31 escampalaboira gnome-session[2796]: Gjs-Message: JS LOG: [FREON] Render all MenuItems
Apr 30 22:38:36 escampalaboira gnome-session[2796]: Gjs-Message: JS LOG: [FREON] Render all MenuItems

UshakovVasilii, 2015-05-01:
Got it, problem related to sensors with the same name `ST3500320AS`.
I'll fix. Thank you

commit 6b01b4bf (2015-05-01): rid full rerander with grouping and temperature sensors count less 3, fix #8
closed 2015-05-01 by UshakovVasilii