UshakovVasilii/gnome-shell-extension-freon #10: Fedora Rawhide now has GNOME Shell 3.17.1

russel, 2015-05-07:
It appears that Fedora Rawhide has updated to GNOME Shell 3.17.1 and so Freon fails to start out of the box. I tried adding "3.17" to the list of versions in metadata.json, but that didn't seem to work – unless I just did something wrong…

UshakovVasilii, 2015-05-12:
I installed Fedora.
Don't work with `3.17`...  Don't work with `3.18`...
Work with `3.17.1`!

You are using not latest gnome version. You need just skip version checking:
`gsettings set org.gnome.shell disable-extension-version-validation "true"`

closed 2015-05-12 by UshakovVasilii

russel, 2015-05-15:
Actually I do have 3.17.1, still doesn't work for me with version checking even though I put 3.17 in the prefs.js file. Or should I have done something other?

russel, 2015-05-15:
Of course switching off version checking, fixes the other extensions as well :-)

PR #13 (2015-10-04): Add support for GNOME 3.18
A very brief test leads me to believe this patch does the trick. Resolves #10 (maybe)
commit a7893c74 (2015-10-04): Add support for GNOME 3.18

A very brief test leads me to believe this patch does the trick. Resolves #10 (maybe)