UshakovVasilii/gnome-shell-extension-freon #11: RPM not updated when fan turns off

ntkoopman, 2015-05-24:
Freon stops updating the RPM display if a fan turns off and will instead keep the last known value.

auipga, 2015-05-24:
can confirm that

UshakovVasilii, 2015-05-24:
What happen with `sensors` output when fan turns off?
Please provide `sensors` output before/after fan turns off/on.
Thanks

auipga, 2015-05-24:
off:
```
i8k-virtual-0
Adapter: Virtual device
Processor Fan:    0 RPM
CPU:            +36.0°C  
SODIMM:         +40.0°C  
Other:          +40.0°C  
GPU:            +44.0°C  

coretemp-isa-0000
Adapter: ISA adapter
Core 0:       +36.0°C  (high = +105.0°C, crit = +105.0°C)
Core 1:       +39.0°C  (high = +105.0°C, crit = +105.0°C)
```
-----------------------
on:
```
i8k-virtual-0
Adapter: Virtual device
Processor Fan: 3035 RPM
CPU:            +58.0°C  
SODIMM:         +52.0°C  
Other:          +52.0°C  
GPU:            +66.0°C  

coretemp-isa-0000
Adapter: ISA adapter
Core 0:       +56.0°C  (high = +105.0°C, crit = +105.0°C)
Core 1:       +59.0°C  (high = +105.0°C, crit = +105.0°C)
```
Freon does not show any fan speed, if the speed is zero on start. The entry is simply missing until in begins to spin

UshakovVasilii, 2015-05-25:
Can anyone confirm fix?

commit 2d4bc11f (2015-05-25): display fans with 0 value, fix #11
commit 12ecebd0 (2015-05-25): display fans with 0 value, fix #11
closed 2015-05-25 by UshakovVasilii