UshakovVasilii/gnome-shell-extension-freon #15: Error displaying Bumblebee Nvidia temperature

re200484, 2015-10-09:
Freon doesn't display the nvidia temperature, it shows me N/A; if i put this in the terminal:
optirun nvidia-smi -q -d TEMPERATURE 
i can see the current temperature.

==============NVSMI LOG==============

Timestamp                           : Fri Oct  9 19:05:09 2015
Driver Version                      : 355.11

Attached GPUs                       : 1
GPU 0000:01:00.0
    Temperature
        GPU Current Temp            : 60 C
        GPU Shutdown Temp           : N/A
        GPU Slowdown Temp           : N/A

UshakovVasilii, 2015-10-09:
"Bumblebee+NVIDIA" display temperature if video card active only.
`optirun nvidia-smi -q -d TEMPERATURE ` command will activate video card, so it is not good approach

closed 2015-10-09 by UshakovVasilii

zhulik, 2017-04-10:
Maybe, it will be better if that behavior will be explicit? Tooltip or any other help message?