UshakovVasilii/gnome-shell-extension-freon #16: Freon spamming syslog

tadfisher, 2015-10-12:
Every 5 seconds, corresponding to the the Freon update interval, Freon logs this message:

```
gnome-session[697]: Gjs-Message: JS LOG: [FREON] Render all MenuItems
```

This really isn't necessary and it ends up spamming the system logs.

UshakovVasilii, 2015-10-13:
This shouldn't happen, please try to use Freon from `additional-logs` branch and provide logs.
Thanks

frosth555, 2016-09-19:
hi, I have the same issue. 
my log:
> 22:22:58 : Gjs-Message: JS LOG: [FREON] Render all MenuItems. Last sensor count is 24, current is 23 Sensors:
22:22:58 : AUXTIN0
22:22:58 : Core 0
22:22:58 : Core 1
22:22:58 : Core 2
22:22:58 : Core 3
22:22:58 : PECI Agent 0
22:22:58 : Physical id 0
22:22:58 : SYSTIN
22:22:58 : temp1
22:22:58 : temp1-1
22:22:58 : temp2
22:22:58 : Crucial_CT256MX100SSD1
22:22:58 : WDC WD30EZRX-00D8PB0
22:22:58 : Average
22:22:58 : Maximum
22:22:58 : temperature-group
22:22:58 : fan1
22:22:58 : fan2
22:22:58 : fan3
22:22:58 : fan4
22:22:58 : fan5
22:23:18 : Gjs-Message: JS LOG: [FREON] Render all MenuItems. Last sensor count is 23, current is 24 Sensors:
22:23:18 : AUXTIN0
22:23:18 : AUXTIN1
22:23:18 : Core 0
22:23:18 : Core 1
22:23:18 : Core 2
22:23:18 : Core 3
22:23:18 : PECI Agent 0
22:23:18 : Physical id 0
22:23:18 : SYSTIN
22:23:18 : temp1
22:23:18 : temp1-1
22:23:18 : temp2
22:23:18 : Crucial_CT256MX100SSD1
22:23:18 : WDC WD30EZRX-00D8PB0
22:23:18 : Average
22:23:18 : Maximum
22:23:18 : temperature-group
22:23:18 : fan1
22:23:18 : fan2
22:23:18 : fan3
22:23:18 : fan4
22:23:18 : fan5
22:23:23 : Gjs-Message: JS LOG: [FREON] Render all MenuItems. Last sensor count is 24, current is 23 Sensors:
22:23:23 : AUXTIN0
22:23:23 : Core 0
22:23:23 : Core 1
22:23:23 : Core 2
22:23:23 : Core 3
22:23:23 : PECI Agent 0
22:23:23 : Physical id 0
22:23:23 : SYSTIN
22:23:23 : temp1
22:23:23 : temp1-1
22:23:23 : temp2
22:23:23 : Crucial_CT256MX100SSD1
22:23:23 : WDC WD30EZRX-00D8PB0
22:23:23 : Average
22:23:23 : Maximum
22:23:23 : temperature-group
22:23:23 : fan1
22:23:23 : fan2
22:23:23 : fan3
22:23:23 : fan4
22:23:23 : fan5

and strange part from sensors(nct6779-isa-0a00):

> CPUTIN:                +127.5°C  (high = +80.0°C, hyst = +75.0°C)  ALARM  sensor = CPU diode
AUXTIN0:                +22.0°C    sensor = thermistor
AUXTIN1:               +127.0°C    sensor = thermistor
AUXTIN2:               -128.0°C    sensor = thermistor
AUXTIN3:                -13.0°C    sensor = thermal diode

hope this helps a little bit.

UshakovVasilii, 2016-09-20:
Thanks!
Currently the following filter:
```
        return s.filter(function(e){
            return e.temp > 0 && e.temp < 115;
        });
```
is used for sensors values.
I'll remove filter.

UshakovVasilii, 2016-09-20:
Of course `nct6779-isa-0a00` values are wrong, but it is `lmsensors` problem

frosth555, 2016-09-20:
Yes, it works thanks a lot.

closed 2016-09-21 by UshakovVasilii