UshakovVasilii/gnome-shell-extension-freon #20: When updating, the extension blocks GNOME Shell

nekohayo, 2015-12-07:
With the default interval of 5 seconds and a bunch of sensors (lm_sensors is installed, and I decided to show 4 CPU core temp sensors + 2 fans on the gnome-shell panel with Freon), if you play music with rhythmbox and type text continuously you will notice that once in a while when freon updates it blocks the shell for a fraction of a second, blocking input and causing a gap in the sound.

nekohayo, 2015-12-07:
Interestingly enough, I think I've noticed the same issue when running the standalone "conky" instead of freon. Might be a clue there.

UshakovVasilii, 2015-12-07:
Hello nekohayo, gap caused by some bad sensor. I have the same issue on my DELL laptop.
You can hear gap after run `sensors` command.
To solve problem you need ignore bad sensor in `sensors` config.

Something like create file `/etc/sensors.d/i8k.conf`
```
chip "i8k-virtual-0"
    ignore fan1
```

closed 2017-06-08 by UshakovVasilii