UshakovVasilii/gnome-shell-extension-freon #33: assertion `CLUTTER_IS_TEXT (self)' failed

satriani-vai, 2016-07-01:
The same [issue like by gnome-shell-extension-sensors](assertion `CLUTTER_IS_TEXT (self)' failed)

```
Jul 01 13:31:15 USER-PC org.gnome.Shell.desktop[977]: (gnome-shell:977): Clutter-CRITICAL **: clutter_text_get_editable: assertion 'CLUTTER_IS_TEXT (self)' failed
Jul 01 13:31:15 USER-PC org.gnome.Shell.desktop[977]: (gnome-shell:977): Clutter-CRITICAL **: clutter_text_get_text: assertion 'CLUTTER_IS_TEXT (self)' failed
Jul 01 13:31:15 USER-PC org.gnome.Shell.desktop[977]: (gnome-shell:977): Clutter-CRITICAL **: clutter_text_set_text: assertion 'CLUTTER_IS_TEXT (self)' failed

$ uname -a
Linux USER-PC 4.6.3-1-ARCH #1 SMP PREEMPT Fri Jun 24 21:19:13 CEST 2016 x86_64 GNU/Linux

$ gnome-shell --version
GNOME Shell 3.20.3
```
gnome-shell-extension-freon was installed from [AUR](https://aur.archlinux.org/packages/gnome-shell-extension-freon-git/) which use a latest source fron git.

terencode, 2016-09-27:
Yes, I have disable right now as it was causing gdm to crash randomly after exiting sleep.

UshakovVasilii, 2017-06-08:
I'm sorry, but I cannot reproduce it and I have no idea what can cause it

UshakovVasilii, 2017-08-12:
I cannot reproduce this problem on gnome 3.24.2, possible problem related to old gnome-shell version

closed 2017-08-12 by UshakovVasilii

BillyCroan, 2022-05-29:
I'm getting it on gnome-shell-42.1-2.fc36.x86_64.