UshakovVasilii/gnome-shell-extension-freon #35: Please run sensors-detect as root won't go away

fiws, 2016-08-29:
hi,

suddenly my temp is gone and it only displays "Please run sensors-detect as root".
I did what it asked for, no change.

this is my sensors output:
```
❯ sensors
acpitz-virtual-0
Adapter: Virtual device
temp1:        +25.0°C  (crit = +107.0°C)

pch_skylake-virtual-0
Adapter: Virtual device
temp1:        +56.0°C  

coretemp-isa-0000
Adapter: ISA adapter
Physical id 0:  +63.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:         +63.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:         +63.0°C  (high = +100.0°C, crit = +100.0°C)

iwlwifi-virtual-0
Adapter: Virtual device
ERROR: Can't get value of subfeature temp1_input: I/O error
temp1:            N/A 
```

fiws, 2016-08-29:
Turning on wifi solves the problem…

new output:
```
acpitz-virtual-0
Adapter: Virtual device
temp1:        +25.0°C  (crit = +107.0°C)

pch_skylake-virtual-0
Adapter: Virtual device
temp1:        +52.0°C  

coretemp-isa-0000
Adapter: ISA adapter
Physical id 0:  +55.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:         +52.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:         +53.0°C  (high = +100.0°C, crit = +100.0°C)

iwlwifi-virtual-0
Adapter: Virtual device
temp1:        +49.0°C
```

i consider this a bug...

UshakovVasilii, 2017-06-08:
Hi,
Looks like sensors command return not zero code. This problem out of this extension. You can skip iwlwifi-virtual-0 chip in lm_sensors configuration.
Thanks

closed 2017-06-08 by UshakovVasilii

svillar, 2017-08-10:
BTW it's as easy as creating a file called `/etc/sensors.d/blacklist-iwlwifi` (for example) and then add the following contents to it:
```
chip "iwlwifi-virtual-0"
	ignore temp1
```

HakimEG, 2019-04-22:
I'm having this issue now, more specifically with hybrid graphics.

The chip(AMD RX550) is N/A when discrete graphics aren't active(ie. not playing games and such)
Which makes FREON return an error.

This forces me to only list my CPU temps, instead of both CPU and GPU, as I otherwise would only get temps when playing games or rendering stuff with the GPU.