UshakovVasilii/gnome-shell-extension-freon #38: support for amdgpu sensor

frosth555, 2016-10-05:
could you add gpu icon for amdgpu driver?

`let isGpuDriver = this._output[i].indexOf("radeon") != -1
				|| this._output[i].indexOf("amdgpu") != -1
                                || this._output[i].indexOf("nouveau") != -1;`
in sensorsUtils.js did the job for me, and now looks better:)

HarlemSquirrel, 2016-11-21:
This sounds like a great idea! Did you make PR?

Veske, 2017-01-31:
Right now it feels that Freon does not have a AMDGPU GPU option at all. Is AMDGPU support even implemented atm?

HarlemSquirrel, 2017-01-31:
I forked from the upstream of this repo and created my own extension here: https://github.com/HarlemSquirrel/gnome-shell-extension-sensory-perception/blob/master/README.md

There I explain a little about customizing the labels.

UshakovVasilii, 2017-06-08:
I use GPU icon for amdgpu, thanks!

closed 2017-06-08 by UshakovVasilii

laichiaheng, 2019-05-19:
@UshakovVasilii Would you add the support for AMDGPU in gnome-shell-extension-freon? The gnome-shell-extension-sensory-perception causes some laggy problem when I watch the videos in windowed mode.

UshakovVasilii, 2019-05-19:
@laichiaheng As I understand current AMD GPU work with lm_sensors https://github.com/UshakovVasilii/gnome-shell-extension-freon/issues/114#issuecomment-491477110

So this problem out of this extension. You can add custom label in lm_sensors for AMD GPU.

laichiaheng, 2019-05-19:
@UshakovVasilii Thanks, it works. I have done it by following the instruction here: https://wiki.archlinux.org/index.php/Lm_sensors#Example_2._Renaming_labels

commit c5fd61fd (2017-06-08): use GPU icon for amdgpu, fix #38