UshakovVasilii/gnome-shell-extension-freon #39: update linuxkernel 4.7.4 to 4.8.4-200.fc24.x86_64 breaks cpu/motherboard sensorinfo

hcbd, 2016-11-01:
Updating the Linuxkernel from 4.7.4 to 4.8.4-200.fd24.x86_64  (Fedora) makes all my CPU temperatures & Fanspeeds dissapear/unavailable in the extension.

Harddrive temperatures are still available, just like before.
GPU temp was and is not used.

the hardware in question:
motherboard: ASUS E3 PRO GAMING V5
CPU: xeon E3-1230v5

gekitsu, 2016-11-29:
i suppose i have a similar issue: ubuntu gnome 16.10, gnome-shell 3.22, kernel 4.8.0-27-generic, and i only get cpu etc. temperatures when wifi is switched on (as in #35). hdd temperature remains available with wifi switched off.

also, switching wifi on, choosing sensors to display in the panel, and switching it back off, looks like it makes the sensors stop updating.

could these issues (kernel version & wifi on/off) be related?

UshakovVasilii, 2017-06-08:
Hi all.
I think these problems related to lm_sensors and out of this extension.
Thanks

closed 2017-06-08 by UshakovVasilii