UshakovVasilii/gnome-shell-extension-freon #41: Clarify documentation on dependencies

terrycloth, 2016-11-20:
I'm preparing an RPM package of your Shell extension for Fedora. I'm a little unclear about the dependencies for reading HDD temperature. Right now, I think you need *both* {hddtemp and gnu-netcat} *or* just {udisks2}. Is that right?

terrycloth, 2016-11-20:
Pending your answer to #40, I can submit updates to the documentation regarding dependencies too :)

terrycloth, 2017-02-22:
This is the dependency list I'm currently using in my RPM package spec.

* gnome-shell >= 3.12
* gnome-shell-extension-common
* gnome-tweak-tool
* lm_sensors
* udisks2

The reslulting package *seems* to work as it should, but I only have my own desktop to test it on, and my hardware's a little old.

[image]

I'm running a local build of Freon 23 on Fedora 25 Workstation x86_64. I have an AMD Athlon(tm) 64 X2 Dual Core Processor 6000+ and an Nvidia GeForce 9600 GT on an MCP61PM-AM motherboard.

UshakovVasilii, 2017-06-08:
Hi, I'm sorry for late reply.
required:
* `lm_sensors`
* `gnome-shell`

optional:
* `udisks2` - backend for S.M.A.R.T. temperature sensors
* `hddtemp` and `gnu-netcat` - disk monitoring alternative
* `nvidea` - NVIDEA card
*  `catalyst`- AMD card
*  `bumblebee` and `nvidea`

closed 2017-06-08 by UshakovVasilii