UshakovVasilii/gnome-shell-extension-freon #44: Gnome crashing when opening folder

Nerdnussbutter, 2016-12-19:
Hello everyone,

When I activate Freon and let it display the temperature icons in the status bar at the top, I can't open any folders in the application overview after some time. This happens shortly after I log in for the first time and disappears temporarely after a reboot. In the first minutes I can open folders, but after that I can't open them any more. A folder would close itself after opening and the application overview starts to lag a ton. Hitting the super key (and thus displaying the desktop) and reopening the application overview fixes the lag, the overview is responsive again. And as I said, disabling the icons fixes the issue (at least for the moment, can't say if it fixes it forever). Can't say if the bug is fixed thanks to not displaying the icons or thanks to refreshing the extension in general either, but the bug won't fix itself if I disable any other extension. 

This is what the /var/log/syslog file says about the error: 

`Dec 19 13:18:24 ubuntu gnome-session[1987]: (gnome-shell:2043): St-CRITICAL **: st_widget_get_theme_node called on the widget [0x2162b60 StBin.app-folder-popup:first-child] which is not in the stage.
Dec 19 13:18:24 ubuntu gnome-session[1987]: (gnome-shell:2043): Gjs-WARNING **: JS ERROR: Error: incorrect pop
Dec 19 13:18:24 ubuntu gnome-session[1987]: popModal@resource:///org/gnome/shell/ui/main.js:444
Dec 19 13:18:24 ubuntu gnome-session[1987]: GrabHelper<._releaseModalGrab@resource:///org/gnome/shell/ui/grabHelper.js:220
Dec 19 13:18:24 ubuntu gnome-session[1987]: wrapper@resource:///org/gnome/gjs/modules/lang.js:178
Dec 19 13:18:24 ubuntu gnome-session[1987]: GrabHelper<.ungrab@resource:///org/gnome/shell/ui/grabHelper.js:267
Dec 19 13:18:24 ubuntu gnome-session[1987]: wrapper@resource:///org/gnome/gjs/modules/lang.js:178
Dec 19 13:18:24 ubuntu gnome-session[1987]: AppFolderPopup<.popdown@resource:///org/gnome/shell/ui/appDisplay.js:1522
Dec 19 13:18:24 ubuntu gnome-session[1987]: wrapper@resource:///org/gnome/gjs/modules/lang.js:178
Dec 19 13:18:24 ubuntu gnome-session[1987]: FolderIcon<._init/<@resource:///org/gnome/shell/ui/appDisplay.js:1244
Dec 19 13:18:24 ubuntu gnome-session[1987]: IconGrid<.destroyAll@resource:///org/gnome/shell/ui/iconGrid.js:682
Dec 19 13:18:24 ubuntu gnome-session[1987]: wrapper@resource:///org/gnome/gjs/modules/lang.js:178
Dec 19 13:18:24 ubuntu gnome-session[1987]: BaseAppView<.removeAll@resource:///org/gnome/shell/ui/appDisplay.js:132
Dec 19 13:18:24 ubuntu gnome-session[1987]: wrapper@resource:///org/gnome/gjs/modules/lang.js:178
Dec 19 13:18:24 ubuntu gnome-session[1987]: _parent@resource:///org/gnome/gjs/modules/lang.js:131
Dec 19 13:18:24 ubuntu gnome-session[1987]: AllView<.removeAll@resource:///org/gnome/shell/ui/appDisplay.js:472
Dec 19 13:18:24 ubuntu gnome-session[1987]: wrapper@resource:///org/gnome/gjs/modules/lang.js:178
Dec 19 13:18:24 ubuntu gnome-session[1987]: BaseAppView<._redisplay@resource:///org/gnome/shell/ui/appDisplay.js:138
Dec 19 13:18:24 ubuntu gnome-session[1987]: wrapper@resource:///org/gnome/gjs/modules/lang.js:178
Dec 19 13:18:24 ubuntu gnome-session[1987]: _runDeferredWork@resource:///org/gnome/shell/ui/main.js:555
Dec 19 13:18:24 ubuntu gnome-session[1987]: _runBeforeRedrawQueue@resource:///org/gnome/shell/ui/main.js:570
Dec 19 13:18:24 ubuntu gnome-session[1987]: _queueBeforeRedraw/<@resource:///org/gnome/shell/ui/main.js:579`

I'm using GNOME Shell 3.18.5 with Ubuntu 16.04 LTS and Kernel Version 4.4.0-53-generic. I noticed the bug last week, around the 15.12.2016, but it could have been around for much longer, since I generally don't use the folders in the overview that much. 
If you need any other data or log files please tell me, I will happily provide them! 
Apart from that great work, I appreciate the extension a lot!

UshakovVasilii, 2017-06-08:
Hi, I'm sorry for late reply.
Please clarify what is `folders in the application overview`? Can you provide screenshot?

Nerdnussbutter, 2017-06-08:
Hi, I didn't realized this thread was still open, I totally forgot it myself. 

What I meant were the folder in the application overview like the one in the attachment:

[image]

But since my post I upgraded to Ubuntu 17.04 with 4.10.2-041002-generic kernel and GNOME Shell 3.24.1 and the issue seems to have solved itself! Maybe the lag was due to some weird interferences. 

As for my part you can mark this as solved.

UshakovVasilii, 2017-06-08:
Oh, good :+1:

closed 2017-06-08 by UshakovVasilii