UshakovVasilii/gnome-shell-extension-freon #45: Trigger some action when a high temperature is reached

Paviluf, 2017-01-19:
Hello, 

In addition of displaying a pop-up notification https://github.com/UshakovVasilii/gnome-shell-extension-freon/issues/12 once the temperature of certain sensors reaches a specified value it will be great if we can trigger some action, like power off the computer, when, for some reason, it's too hot.

Thanks !

ColinHarrington, 2017-10-25:
Adding an arbitrary/configurable action might be nice.

I don't think it's a gnome-shell extension's job to shut down a hot machine.  You should be using some other daemon to ensure your hardware survives - not an extension to a graphical shell.

AlexWayfer, 2017-10-25:
Maybe alerts? Dialog window (with sound maybe) about dangerous level.