UshakovVasilii/gnome-shell-extension-freon #46: Feature Request (4 graphics tempereture in the panel)

cyberalex4life, 2017-02-16:
Hello,
I've been using your extension for some time now. I use and intel/nvidia optimus laptop with bumblebee and I would enjoy keeping Bumblebee+NVidia temperature selected for showing in the panel all the time. However, it kind of bugs me that when nvidia is not active the N/A appears instead. My idea would be to just hide the icon symbol together with the N/A text when it's there's no bumblebee-nvidia It would make more sense to me, what do you say?

UshakovVasilii, 2017-06-08:
Hi, I'm sorry for late reply.
We cannot hide N/A and icon if only one indicator is shown in top panel. And if more one indicators are shown in top panel then other indicators will blink (bumblebee indicator will permanently hide & show).
Thanks

closed 2017-06-08 by UshakovVasilii

cyberalex4life, 2017-06-08:
I understand.