UshakovVasilii/gnome-shell-extension-freon #47: Bumblebee+NVIDIA temperature shows N/A

dosfanat, 2017-02-17:
Hi! I think i found an incompatibility issue. Or a bug (much less probable)

Bumblebee+NVIDIA temperature shows N/A even when i am running something opengl-ish through primusrun or optirun. The result is simply the same - no value, omly N/A.

Laptop model: Acer 5750G
My distro: Debian 8.7 jessie
NVIDIA proprietary driver 340.101
Gnome Shell 3,14.4-1-deb8u
If needed - additional info will be quickly provided.

dimmuboy, 2017-02-24:
old thing
[https://github.com/UshakovVasilii/gnome-shell-extension-freon/issues/15](url)

UshakovVasilii, 2017-06-08:
Hi, I'm sorry for late reply.
Is `optirun nvidia-smi -q -d TEMPERATURE` return any info?

dimmuboy, 2017-06-09:
In my case this command return

> NVIDIA-SMI has failed because it couldn't communicate with the NVIDIA driver. Make sure that the latest NVIDIA driver is installed and running.

UshakovVasilii, 2017-06-09:
It is reason of problem.
Problem out of gnome extension, so I close the ticket

closed 2017-06-09 by UshakovVasilii