UshakovVasilii/gnome-shell-extension-freon #49: AMDGPU support

Veske, 2017-03-11:
Looks like maintainer is not active? Made my own fork with AMDGPU support. https://github.com/Veske/gnome-shell-extension-freon

If maintainer comes back, please pull changes in.

UshakovVasilii, 2017-06-08:
Hi, I'm sorry for so late reply.
Is file `amdGpuSensors.conf` necessary for AMD gpu support? We cannot pack this file into zip archive for https://extensions.gnome.org

UshakovVasilii, 2017-06-08:
Ah, we can monitor AMD from sensors utility, so we not need any special configuration and code modification. Am I wrong?

closed 2017-06-08 by UshakovVasilii

Veske, 2017-06-08:
Yes, sensors conf would be there just for convenience. You can change the format of sensor output with it.

UshakovVasilii, 2017-06-09:
So #38 enough for AMDGPU support.
Thanks!