UshakovVasilii/gnome-shell-extension-freon #54: Missing CPU sensors with kernel 4.11.3

fchiacchiaretta, 2017-06-09:
Hi,
I recently updated to kernel 4.11.3 on Fedora 25 and since then CPU sensors are missing.
I run sensors-detect and this is the output of sensors:
```
iwlwifi-virtual-0
Adapter: Virtual device
ERROR: Can't get value of subfeature temp1_input: I/O error
temp1:            N/A  

acpitz-virtual-0
Adapter: Virtual device
temp1:        +45.0°C  (crit = +128.0°C)
temp2:         +0.0°C  (crit = +128.0°C)
temp3:        +39.0°C  (crit = +128.0°C)
temp4:        +41.0°C  (crit = +128.0°C)
temp5:        +28.0°C  (crit = +128.0°C)
temp6:       +127.0°C  (crit = +128.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +45.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +43.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +44.0°C  (high = +100.0°C, crit = +100.0°C)

pch_skylake-virtual-0
Adapter: Virtual device
temp1:        +43.5°C
```
I running Fedora package gnome-shell-extension-freon-23-4.fc25.noarch for Freon on HP Probook 440 G3.

Is there any log I can check to understand the issue? I tried using looking glass (Alt-F2 lg) without luck.
Thanks.

fchiacchiaretta, 2017-06-09:
I found useful info in issue #39 , enabling Wi-Fi I found my CPU sensors listed again.
As you can see from the sensors output above, I have an ERROR line:

`ERROR: Can't get value of subfeature temp1_input: I/O error`

If I enable Wi-Fi this line disappears.
Does this extension parse the output of sensors command? If so, can you workaround this ERROR within parsing?

Thanks.

UshakovVasilii, 2017-06-11:
Hi,
You can ignore `temp1` for `iwlwifi-virtual-0` chip https://github.com/UshakovVasilii/gnome-shell-extension-freon/wiki/FAQ
Thanks

fchiacchiaretta, 2017-06-12:
Sorry, I should have searched better in other issues and FAQ.
Ignoring it in config fixed the issue.

Thanks

closed 2017-06-12 by fchiacchiaretta