UshakovVasilii/gnome-shell-extension-freon #56: No longer detects sensor output

rwbaskette, 2017-08-05:
I believe "something" changed in the last couple weeks after updating my Ubuntu Gnome 17.04 system.
I'm being prompted to run sensors-detect again from the menu.
Could the output format of sensors have changed?

```lsb_release -a && echo --------- && uname -a && echo -------- && sensors 
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 17.04
Release:	17.04
Codename:	zesty
---------
Linux sigma7 4.10.0-30-generic #34-Ubuntu SMP Mon Jul 31 19:38:17 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
--------
pch_skylake-virtual-0
Adapter: Virtual device
temp1:        +42.5°C  

acpitz-virtual-0
Adapter: Virtual device
temp1:        +45.0°C  (crit = +103.0°C)

iwlwifi-virtual-0
Adapter: Virtual device
ERROR: Can't get value of subfeature temp1_input: I/O error
temp1:            N/A  

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +50.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +44.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +50.0°C  (high = +100.0°C, crit = +100.0°C)
```

UshakovVasilii, 2017-08-06:
Hi,
You got error in `sensors` utility: `ERROR: Can't get value of subfeature temp1_input: I/O error`.
After that `sensors` return not `0` exit code.
Solution here https://github.com/UshakovVasilii/gnome-shell-extension-freon/issues/54#issuecomment-307608998

rwbaskette, 2017-08-06:
thanks for the quick response!

closed 2017-08-06 by rwbaskette

rwbaskette, 2017-08-06:
Is there a way to ignore the error return value? 

I only press because I have a laptop and when I'm at my desk, i turn off wifi to force all my traffic out the ethernet port. Turning off wifi causes sensors to return an error because it can't read a configured sensor.

I'm not a gnome/glib/etc dev, but I'm looking at commandLintUtil.js and the docs scratching my head as to why the child_watch_add would care about the return code of the sensors binary. The output of sensors still writes to stdout even if there is a "sensor error".

Any thoughts?

Thanks!

UshakovVasilii, 2017-08-07:
Ah, exit code is not necessary. I tried to use your output, but extension still work on my laptop.

It is difficult to fix problem without reproducing. I think need to modify "sensorsUtil.js#_parseSensorsOutput"  to skip error line.

You can use clone repository, fix problem and create pull request. :-)

* use `deploy.sh` to deploy extension locally
* `Alt+f2`, `r` to restart gnome-shell
* `global.log("Hello World")` - send logs to `journalctl`
* `journalctl -f` to watch logs

Thank you!