UshakovVasilii/gnome-shell-extension-freon #57: Disk IOPS report

rickysarraf, 2017-08-07:
Thank you for maintaining this extension. It would be nice if the extension could also monitor and display disk iops in its report. Disk activity is one of the most common task and tracking it through this extension would be a nice enhancement.

UshakovVasilii, 2017-08-07:
Hi,
https://extensions.gnome.org/extension/120/system-monitor/ extension can show IO read/write MiB/s. Is it not enough?
I'm not sure what iops monitoring should be task for this extension.
Thanks

rickysarraf, 2017-08-07:
Yes. I'm already using that extension. I just thought it'd be a good addition to this extension too. But yes, if this extension's scope is limited to temperature sensor data, then adding that would be beyond the targeted scope.

closed 2017-08-07 by UshakovVasilii