UshakovVasilii/gnome-shell-extension-freon #58: Too many Nvidia cards showing

stevef9432203, 2017-08-12:
pita-24  Sat Aug 12 14:26:59 2017
[0] GeForce GTX 1080 Ti | 46'C,   1 % |   377 / 11172 MB |
[1] GeForce GTX 1070    | 40'C,   0 % |     2 /  8114 MB |
[image]

UshakovVasilii, 2017-08-13:
Please provide the following commands output:
```
nvidia-settings -q gpucoretemp -t
nvidia-settings -q gpucoretemp
nvidia-settings -q gpus
```

UshakovVasilii, 2017-08-13:
```
$ nvidia-settings -q gpucoretemp -t
79
79
58
$ nvidia-settings -q gpucoretemp

  Attribute 'GPUCoreTemp' (pita-24:1.0): 78.
    'GPUCoreTemp' is an integer attribute.
    'GPUCoreTemp' is a read-only attribute.
    'GPUCoreTemp' can use the following target types: X Screen, GPU.
  Attribute 'GPUCoreTemp' (pita-24:1[gpu:0]): 78.
    'GPUCoreTemp' is an integer attribute.
    'GPUCoreTemp' is a read-only attribute.
    'GPUCoreTemp' can use the following target types: X Screen, GPU.
  Attribute 'GPUCoreTemp' (pita-24:1[gpu:1]): 58.
    'GPUCoreTemp' is an integer attribute.
    'GPUCoreTemp' is a read-only attribute.
    'GPUCoreTemp' can use the following target types: X Screen, GPU.

$ nvidia-settings -q gpus

2 GPUs on pita-24:1

    [0] pita-24:1[gpu:0] (GeForce GTX 1080 Ti)

      Has the following names:
        GPU-0
        GPU-4421d1e1-42a0-3483-2583-fa012499e659

    [1] pita-24:1[gpu:1] (GeForce GTX 1070)

      Has the following names:
        GPU-1
        GPU-a8b9f8a1-11f9-7cea-71f2-0c23c628c516
```

closed 2017-08-13 by UshakovVasilii