UshakovVasilii/gnome-shell-extension-freon #59: No temperatures shown in bar with no discrete GPU

DuBistKomisch, 2017-08-18:
From the screenshots I've seen it's meant to show some sort of average temperature next to the icon in the bar, but all I see is the first icon:

[image]

I use VFIO / GPU passthrough to a windows VM, and run Arch as a host on the integrated GPU, so I don't expect to be able to see the discrete GPU temperature. I've selected "None" for "Video Card Temperature Utility" so not sure what it's trying to do.

UshakovVasilii, 2017-08-19:
You can click to icon and select indicator to show

DuBistKomisch, 2017-08-19:
Haha, so you can, oops. Feel free to close this.

Maybe should be mentioned explicitly somewhere? Or default to average?

UshakovVasilii, 2017-08-19:
I create issue #60
Thanks!

closed 2017-08-19 by UshakovVasilii