UshakovVasilii/gnome-shell-extension-freon #61: nvme support

EUA, 2017-09-05:
Hi, I cannot see my NVME device on freon.
I use uDisk2 as HDD/SDD temperature utility
I can saw my unit using udis also.
Is this problem due freon or dbus/udisk?
Thx

```
[root@TriQuad death]# udisksctl status
MODEL                     REVISION  SERIAL               DEVICE
--------------------------------------------------------------------------
-                                   S346NY0HB08149       nvme0n1 
ST340014A                 3.54      3JV9A6SB             sda     
SAMSUNG HD250HJ           FH100-06  S0URJDWQ605033       sdb     
SAMSUNG HD250HJ           FH100-06  S189J9DQ202933       sdc     
SAMSUNG SP2504C           VT100-33  S09QJ1IL204093       sdd     
Maxtor 7L250S0            BANC1G10  L50QT75H             sde  
```

UshakovVasilii, 2017-09-05:
Hi, please provide file with `udisksctl dump | cat` output

EUA, 2017-09-06:
Himm. It looks like Udisk doesn't show Smart features for nvme.

[udiskctldump.txt](https://github.com/UshakovVasilii/gnome-shell-extension-freon/files/1279418/udiskctldump.txt)

 I can see thru smartctl instead.
```
[death@TriQuad ~]$ sudo smartctl -a /dev/nvme0n1
[sudo] password for death: 
smartctl 6.5 2016-05-07 r4318 [x86_64-linux-4.13.0-1-hardened] (local build)
Copyright (C) 2002-16, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Number:                       SAMSUNG MZVPW256HEGL-00000
Serial Number:                      Decided to mask this area :D
Firmware Version:                   CXZ7300Q
PCI Vendor/Subsystem ID:            0x144d
IEEE OUI Identifier:                0x002538
Total NVM Capacity:                 256.060.514.304 [256 GB]
Unallocated NVM Capacity:           0
Controller ID:                      2
Number of Namespaces:               1
Namespace 1 Size/Capacity:          256.060.514.304 [256 GB]
Namespace 1 Formatted LBA Size:     512
Local Time is:                      Wed Sep  6 04:05:41 2017 +03
Firmware Updates (0x16):            3 Slots, no Reset required
Optional Admin Commands (0x0017):   Security Format Frmw_DL *Other*
Optional NVM Commands (0x001f):     Comp Wr_Unc DS_Mngmt Wr_Zero Sav/Sel_Feat
Warning  Comp. Temp. Threshold:     70 Celsius
Critical Comp. Temp. Threshold:     73 Celsius

Supported Power States
St Op     Max   Active     Idle   RL RT WL WT  Ent_Lat  Ex_Lat
 0 +     6.80W       -        -    0  0  0  0        0       0
 1 +     5.60W       -        -    1  1  1  1        0       0
 2 +     4.40W       -        -    2  2  2  2        0       0
 3 -   0.0400W       -        -    3  3  3  3      210    1500
 4 -   0.0050W       -        -    4  4  4  4     2200    6000

Supported LBA Sizes (NSID 0x1)
Id Fmt  Data  Metadt  Rel_Perf
 0 +     512       0         0

=== START OF SMART DATA SECTION ===
SMART overall-health self-assessment test result: PASSED

SMART/Health Information (NVMe Log 0x02, NSID 0x1)
Critical Warning:                   0x00
Temperature:                        39 Celsius
Available Spare:                    100%
Available Spare Threshold:          10%
Percentage Used:                    2%
Data Units Read:                    15.793.258 [8,08 TB]
Data Units Written:                 13.182.260 [6,74 TB]
Host Read Commands:                 601.940.133
Host Write Commands:                142.173.994
Controller Busy Time:               1.005
Power Cycles:                       630
Power On Hours:                     728
Unsafe Shutdowns:                   296
Media and Data Integrity Errors:    0
Error Information Log Entries:      82
Warning  Comp. Temperature Time:    0
Critical Comp. Temperature Time:    0
Temperature Sensor 1:               39 Celsius
Temperature Sensor 2:               48 Celsius

Error Information (NVMe Log 0x01, max 64 entries)
Num   ErrCount  SQId   CmdId  Status  PELoc          LBA  NSID    VS
  0         82     0  0x0018  0x4004  0x02c            0     0     -
  1         81     0  0x0017  0x4004  0x02c            0     0     -
  2         80     0  0x0018  0x4004  0x02c            0     0     -
  3         79     0  0x0017  0x4004  0x02c            0     0     -
  4         78     0  0x0016  0x4004  0x02c            0     0     -
  5         77     0  0x0015  0x4004  0x02c            0     0     -
  6         76     0  0x0016  0x4004  0x02c            0     0     -
  7         75     0  0x0015  0x4004  0x02c            0     0     -
  8         74     0  0x0016  0x4004  0x02c            0     0     -
  9         73     0  0x0015  0x4004  0x02c            0     0     -
 10         72     0  0x0018  0x4004  0x02c            0     0     -
 11         71     0  0x0017  0x4004  0x02c            0     0     -
 12         70     0  0x0018  0x4004  0x02c            0     0     -
 13         69     0  0x0017  0x4004  0x02c            0     0     -
 14         68     0  0x0018  0x4004  0x02c            0     0     -
 15         67     0  0x0017  0x4004  0x02c            0     0     -
... (48 entries not shown)
```

UshakovVasilii, 2017-09-06:
Looks like driver not so good. S.M.A.R.T. not supported, smartcl output has no
```
SMART support is: Available - device has SMART capability.
SMART support is: Enabled
```
fields.  `SMART DATA SECTION`  has no SMART Attribute `Temperature_Celsius`.
I don't know how monitor temperature of this device correctly. Do any utility show temperature of this device?

EUA, 2017-09-08:
No I just use smartctl for it.
`[root@TriQuad death]# smartctl -a /dev/nvme0n1 | grep Temperature:
Temperature:                        43 Celsius`

Another utility:
```
[root@TriQuad death]# nvme smart-log /dev/nvme0
Smart Log for NVME device:nvme0 namespace-id:ffffffff
critical_warning                    : 0
temperature                         : 43 C
available_spare                     : 100%
available_spare_threshold           : 10%
percentage_used                     : 2%
data_units_read                     : 16.159.812
data_units_written                  : 13.461.365
host_read_commands                  : 604.981.506
host_write_commands                 : 143.471.824
controller_busy_time                : 1.021
power_cycles                        : 639
power_on_hours                      : 735
unsafe_shutdowns                    : 299
media_errors                        : 0
num_err_log_entries                 : 82
Warning Temperature Time            : 0
Critical Composite Temperature Time : 0
Temperature Sensor 1                : 43 C
Temperature Sensor 2                : 50 C
```

UshakovVasilii, 2017-09-09:
Looks like problem related to driver and out of gnome-shell-extension.
But I create enhancement #62

closed 2017-09-12 by UshakovVasilii

EUA, 2018-06-30:
Issue is due UDisk2 definitely at this bug: https://github.com/storaged-project/udisks/issues/386

ibnYusrat, 2020-10-05:
Any update on this? This is still an issue with me. I can see my disk in `smartctl` but can't see it inside freon. Freon is my favorite gnome extension and this small issue is a bit annoying..

EUA, 2020-10-31:
@ibnYusrat UDisk2 able to show my controller sensors properly.