UshakovVasilii/gnome-shell-extension-freon #63: Session freezed when Nvidia is disabled

junquera, 2017-10-06:
If I set that Nvidia temperature to be shown and I disable the card (running `prime-select intel`), when I reboot, it gets freezed and I can no longer login. If I do not disable the extension, its impossible to login.

I think that it could be a loop or some error with `GLib.spawn_command_line_sync` line (in `nvidiaUtil.js`). I remember once I wrote a code that got freezed when a program was executed and it only writes in `stderror` because it stills waiting till receiving somethin, but I have no idea about Gnome Shell extensions programming, and I'm not sure that if it works exactly like java...

UshakovVasilii, 2017-10-08:
Oh, I didn't care about `prime-select`.
Can you try to do the following:
1. disable freon extension
2. select intel card `prime-select intel`
3. reboot
4. execute `nvidia-settings -q  gpus`

I want to know result.

junquera, 2017-10-08:
Hi! This is the result:
```
junquera@opa:~$ nvidia-settings -q gpus

ERROR: Unable to find display on any available system

ERROR: Unable to find display on any available system
```

skobkin, 2017-10-08:
Had the same problem. Reported it using [Gnome Extension Error Reporting form](https://extensions.gnome.org/errors/report/841) with details.

UshakovVasilii, 2017-10-09:
Thanks!
Can you also do the following with disabled card:
1. `gjs`
2.
``` js
const GLib = imports.gi.GLib;
GLib.spawn_command_line_sync('nvidia-settings -q gpus');
GLib.spawn_command_line_sync('nvidia-settings -q gpucoretemp -t');
```

junquera, 2017-10-10:
This is the result:
```
gjs> const GLib = imports.gi.GLib;
gjs> GLib.spawn_command_line_sync('nvidia-settings -q gpus');
true,
,
ERROR: Unable to find display on any available system

ERROR: Unable to find display on any available system

,0
gjs> GLib.spawn_command_line_sync('nvidia-settings -q gpucoretemp -t');
true,,
ERROR: Unable to find display on any available system

ERROR: Unable to find display on any available system

,256
```

UshakovVasilii, 2017-10-12:
I'm sorry but I cannot find problem reason, I have no device with that problem..

I think problem related to this code https://github.com/UshakovVasilii/gnome-shell-extension-freon/blob/master/freon%40UshakovVasilii_Github.yahoo.com/commandLineUtil.js#L14-L39
You can use `global.log('...')` to output logs to `journalctl`, https://github.com/UshakovVasilii/gnome-shell-extension-freon/blob/master/deploy.sh script to put extension in proper place and `Alt+F2` `r` to restart gnome-shell

I'll be glad to see pull request :-)

skobkin, 2017-10-12:
@UshakovVasilii 
```
$ gjs        
gjs> const GLib = imports.gi.GLib;
gjs> GLib.spawn_command_line_sync('nvidia-settings -q gpus');
true,
,
ERROR: Unable to find display on any available system

ERROR: Unable to find display on any available system

,0
gjs> GLib.spawn_command_line_sync('nvidia-settings -q gpucoretemp -t');
true,,
ERROR: Unable to find display on any available system

ERROR: Unable to find display on any available system

,256
```

I'm using Wayland:

```
$ loginctl show-session 1 | fgrep Type
Type=wayland
```

And also I didn't load nvidia module:

```
$ lsmod 
Module                  Size  Used by
i915                 1277952  39
intel_gtt              20480  1 i915
i2c_algo_bit           16384  1 i915
drm_kms_helper        110592  1 i915
syscopyarea            16384  1 drm_kms_helper
sysfillrect            16384  1 drm_kms_helper
sysimgblt              16384  1 drm_kms_helper
fb_sys_fops            16384  1 drm_kms_helper
drm                   262144  28 i915,drm_kms_helper
video                  36864  1 i915
```

Yes, it's full list of loaded modules because I'm using almost monolith kernel.

UshakovVasilii, 2017-10-12:
Thanks, I thought `spawn_command_line_sync` will cause system freeze, but problem out of this function. Unfortunately I cannot find problem.  https://github.com/UshakovVasilii/gnome-shell-extension-freon/issues/63#issuecomment-336041215

junquera, 2017-10-12:
I've been testing, and the log after this line doesn't appear in my `journalctl`. I wrote:

``` js
global.log("[TEST] Before");
let [res, out] = GLib.spawn_command_line_sync(path + " -q gpus")
global.log("[TEST] After");
```

and it just appears `[TEST] Before`.

Also see that `nvidia-settings` is called at `commandLineUtil.js`:

``` js
            let [exit, pid, stdinFd, stdoutFd, stderrFd] =
                GLib.spawn_async_with_pipes(null, /* cwd */
                                            this._argv, /* args */
                                            null, /* env */
                                            GLib.SpawnFlags.DO_NOT_REAP_CHILD,
                                            null /* child_setup */);
```

But this command works too running with `gjs`. Could it be the "user level" where extensions are executed?

P.S. I'll be glad to make the pull request, of course! But i'll nedd some days to find the problem :sweat_smile:

UshakovVasilii, 2017-10-13:
But if you do the following steps:
* disable freon extension (NVIDIA option selected) 
* select intel card `prime-select intel`
* reboot
* enable extension

Will everything be OK? If so then possible we need to add timeout for Nvidia monitoring start (Of cause it is not good code, but if this problem can be related to Nvidia driver and we have no other option).

Need to modify `extension.js, line 39`
```
        Mainloop.timeout_add_seconds(10, Lang.bind(this, function (){
            this._initGpuUtility();
            // disable timer
            return false;
        }));
```

junquera, 2017-10-13:
No, also crashes when I enable the extension after the session hash loaded...

UshakovVasilii, 2017-10-14:
@junquera I reset https://github.com/UshakovVasilii/gnome-shell-extension-freon/pull/64#issuecomment-336615370
Extension is not started after merge, syntax error, missing `)` and `}`

UshakovVasilii, 2017-10-15:
Thanks @junquera! I published new version to extensions.gnome.org, it will be available after review.

closed 2017-10-15 by UshakovVasilii