UshakovVasilii/gnome-shell-extension-freon #66: Fails to gather values when lm-sensors returns some errors

baco, 2017-10-17:
On my PC ya have two groups of sensors detected by lm-sensors: `coretemp-isa-0000` and `iwlwifi-virtual-0`. The `coretemp-isa-0000` group is working always but the `iwlwifi-virtual-0` device is only enabled when the WiFi adapter is on.

When the WiFi adapter is off, the output of the `sensors` command is something like this:

    $ sensors
    iwlwifi-virtual-0
    Adapter: Virtual device
    ERROR: Can't get value of subfeature temp1_input: I/O error
    temp1:            N/A  

    coretemp-isa-0000
    Adapter: ISA adapter
    Core 0:       +51.0°C  (high = +90.0°C, crit = +90.0°C)
    Core 1:       +51.0°C  (high = +90.0°C, crit = +90.0°C)
    Core 2:       +52.0°C  (high = +90.0°C, crit = +90.0°C)
    Core 3:       +47.0°C  (high = +90.0°C, crit = +90.0°C)

When that happens Freon fails to report all the other good values that I would like to be informed of.

UshakovVasilii, 2017-10-19:
Hello, please check #56

baco, 2017-10-20:
Ok, that solution is **very** wrong. I don't intend to ignore that sensor's value forever.

As I said, the sensor has a valid value when the WiFi adapter is on and I care about that value as for any other sensors' values. The issue is that I constantly turn on and off the WiFi adapter according to my needs on battery discharge, but when the adapter is on I want to know it's temperature value.

UshakovVasilii, 2017-10-20:
I think `lm_sensors` shouldn't return error for turned off adapter. Anyway I couldn't fix this problem you are free to fix problem on extension side and create pull request. :-)

UshakovVasilii, 2019-05-17:
Possible fixed by #114

arigit, 2019-05-17:
For this one you will need another tweak,

```
....
            let lineRemoved = this._output.filter(l => l.trim() !== ',').join('\n');
            let errorRemoved = lineRemoved.replace(/ERROR.*Can't read/, "");
            let errorRemoved2 = errorRemoved.replace(/ERROR.*I/O error/, "");
            data = JSON.parse(errorRemoved2);

```

UshakovVasilii, 2019-05-17:
OK, thanks

closed 2019-05-17 by UshakovVasilii