UshakovVasilii/gnome-shell-extension-freon #73: Only temperatures are shown

amivaleo, 2017-11-24:
Hi!
I installed lm_sensors on my ubuntu 17.10. Now I can see only temperatures. I would like to see the fan speed and the voltage too. How do I get them?

Thank you. :)
[image]

UshakovVasilii, 2017-11-24:
Hello!
Extension use `lm_sensors`, you can execute `sensors` command to check extension data source. 
You can use ` # yes | sensors-detect` to configure sensors. Good wiki page https://wiki.archlinux.org/index.php/lm_sensors

Also check Voltage and Fan speed checkboxes on extension settings.

amivaleo, 2017-11-26:
I checked that page. I gave a "sensors-detect" command and answered "yes" to every check and scan, so that now everything should be on after that. Still, I can only see temperatures.
I'm going to install some software with a gui, so I can make a comparison and check if the fan speed (for example) is shown there.

amivaleo, 2017-11-26:
Ok, I installed hardinfo and... I can only see temperatures there! o.o
Do you know if I have to install some other package in order to see voltages and fan speed?

UshakovVasilii, 2017-11-27:
As I understand `sensors` command is not show Fan speed and Voltage. So problem related to drivers or hardware.

closed 2017-11-27 by UshakovVasilii