UshakovVasilii/gnome-shell-extension-freon #78: GPU Temp erroneously shown when GPU is offline

rahulkolasseri, 2018-01-02:
When the GPU is offline, the sensor goes offline as well and Freon shows a useless reading instead of a Blank/OFF value. If showing average sensor temperature, it takes this erroneous value into account as well instead of discarding it, making the average value shown completely wrong too.

Eg.
[image]
(Fan, CPU, GPU offline, HDD)

[image]
(Fan, CPU, GPU online, HDD)

UshakovVasilii, 2018-01-20:
> When the GPU is offline, the sensor goes offline as well and Freon shows a useless reading instead of a Blank/OFF value. 

As I understand you use Bumblebee+NVIDIA for GPU monitoring.
When GPU online file `/tmp/.X:8` should exist but looks like you don't have this file and GPU monitoring command `optirun nvidia-smi -q -d TEMPERATURE` executed and got wrong value. Please check `/tmp/.X:8` file and `/etc/bumblebee/bumblebee.conf`#VirtualDisplay= value.

> If showing average sensor temperature, it takes this erroneous value into account as well instead of discarding it, making the average value shown completely wrong too.

It is problem, currently temperatures from `lm_sensors` only involve in average temperature calculation. I will fix it later #81

rahulkolasseri, 2018-01-31:
Actually I am using AMD, not Nvidia. And I have selected "Catalyst" under "Video card temp. utility", and it works when the card is online although I don't have catalyst installed, am using the AMDGPU drivers instead.

UshakovVasilii, 2018-02-04:
ok, please provide `aticonfig --odgt` output with on/off Catalyst

UshakovVasilii, 2019-04-04:
deprecated

closed 2019-04-04 by UshakovVasilii