UshakovVasilii/gnome-shell-extension-freon #79: GNOME Shell crashes, window title bars disappear, then the Shell resets or logs out

terrycloth, 2018-01-13:
### Symptoms

I'm not sure what the trigger is here. GNOME Shell freezes randomly. I can usually still move the mouse around, but nothing responds to mouse clicks or keyboard presses. Oftentimes, after a few seconds, all window titlebars vanish, and things stay frozen. This can last for up to a minute, before the Shell seems to reload: The screen flashes, and then everything is back to normal, though the windows in the current workspace may be out of order. The windows that have focus before and after the crash may not be the same.

A few times, instead of allowing me to resume what I was doing, I got kicked to the GDM login screen, and any unsaved work was lost.

The freeze can happen several times a day, and has been an ongoing issue at least since GNOME 3.24. It was so bad that for a while I started using Cinnamon instead... but I like GNOME better, and I feel like investigating.

### Narrowing down the source of the problem

Since GNOME Shell is what's crashing, I originally reported this issue against the [GNOME Shell Bugzilla](https://bugzilla.gnome.org/show_bug.cgi?id=791055), but more recently I thought to try testing my Shell *extensions*.

I started by disabling all of them -- suddenly the Shell crash stopped happening -- and I gradually reenabled extensions one by one, waiting a few days in between each. The crash did not happen again until I finally got to Freon, and then Shell started crashing several times a day again.

Just to make sure, I turned Freon back off: no crashes for several days. And then I turned it back on: several crashes in one day again.

When I first started to really investigate the issue, I installed the debuginfo packages for gjs, gnome-shell, and gdm, then waited for the crash to happen again. ABRT generated [this big (37 MB) tarball of debugging info](https://andrew.tosk.in/tmp/gnome-shell-killed-by-sigsegv--ccpp-2017-11-30-17:28:22.954305-2938.tar.gz), but I don't know if this has the sort of information *you* would need. Let me know what other information I can provide that might help.

### My system setup

* Fedora 27 Workstation x86_64
* GNOME Shell 3.26.2 (running in an Xorg session)
* GDM 3.26.2.1
* GNOME Shell extension Freon 31
* lm_sensors 3.4.0-8.fc27
* Linux kernel 4.14.13-300.fc27
* kinda old desktop hardware:
    * AMD Phenom(tm) ii x4 955 processor
    * GeForce GTX 560 Ti/PCIe/SSE2 graphics with proprietary driver from RPM Fusion:
        * akmod-nvidia 387.34-1.fc27
        * xorg-x11-drv-nvidia 387.34-2.fc27
    * 250 GB hard disk drive
    * 8 GB DDR3 RAM
* [image]

I can easily install other debug packages if needed, though if there's some other better process for generating these crash reports, I may need you to walk me through the steps needed.

UshakovVasilii, 2018-01-13:
Hello, I think it related to #74.
I updated master, but didn't upload extension on extensions.gnome.org yet

terrycloth, 2018-01-15:
Okay, I'll be sure to test when version 33 is released.

UshakovVasilii, 2018-01-20:
version 33 under review on https://extensions.gnome.org

closed 2018-01-20 by UshakovVasilii