UshakovVasilii/gnome-shell-extension-freon #82: CPU temp not showing

rawsh, 2018-01-22:
Ryzen 1800x. 

[image]

UshakovVasilii, 2018-01-23:
Please provide `sensors` output

rawsh, 2018-01-23:
@UshakovVasilii It's a very strange bug. The fix is to enable wifi. With wifi disabled it does not show cpu temp or voltages.

```
sensors
nct6792-isa-0290
Adapter: ISA adapter
in0:                    +0.38 V  (min =  +0.00 V, max =  +1.74 V)
in1:                    +0.00 V  (min =  +0.00 V, max =  +0.00 V)
in2:                    +3.28 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                    +3.28 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                    +0.00 V  (min =  +0.00 V, max =  +0.00 V)
in5:                    +0.00 V  (min =  +0.00 V, max =  +0.00 V)
in6:                    +0.84 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                    +3.46 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                    +3.28 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                    +0.00 V  (min =  +0.00 V, max =  +0.00 V)
in10:                   +0.12 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                   +0.13 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                   +1.83 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                   +1.67 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                   +0.00 V  (min =  +0.00 V, max =  +0.00 V)
fan1:                  1440 RPM  (min =    0 RPM)
fan2:                     0 RPM  (min =    0 RPM)
fan3:                  1310 RPM  (min =    0 RPM)
fan4:                     0 RPM  (min =    0 RPM)
fan5:                     0 RPM  (min =    0 RPM)
SYSTIN:                +120.0°C  (high =  +0.0°C, hyst =  +0.0°C)  sensor = thermistor
CPUTIN:                 +27.5°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                +35.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM  sensor = thermistor
AUXTIN1:               +109.0°C    sensor = thermistor
AUXTIN2:               +110.0°C    sensor = thermistor
AUXTIN3:               -128.0°C    sensor = thermistor
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C  
PCH_CHIP_TEMP:           +0.0°C  
PCH_CPU_TEMP:            +0.0°C  
PCH_MCH_TEMP:            +0.0°C  
intrusion0:            OK
intrusion1:            ALARM
beep_enable:           disabled

iwlwifi-virtual-0
Adapter: Virtual device
temp1:        +34.0°C  
```

UshakovVasilii, 2018-01-25:
I think it is same behavior as #66

closed 2019-04-04 by UshakovVasilii