UshakovVasilii/gnome-shell-extension-freon #85: Average is too low because of zero values

AlexWayfer, 2018-02-16:
Hello!

What's wrong with sensors where tempature is zero?

[image]

UshakovVasilii, 2018-02-21:
Hello, this problem related to [lm_sensors](https://wiki.archlinux.org/index.php/lm_sensors).
You can reconfigure sensors or skip broken values. You can use `sensors` command to check it.

AlexWayfer, 2018-02-21:
> You can reconfigure sensors `sensors-detect --auto`

It didn't help :(

> or skip broken values

But I don't know where are `AUXTIN`, `CPUTIN`, etc. in selection dialogs:

```
# sensors-detect revision $Revision$
# Board: ASUSTeK COMPUTER INC. SABERTOOTH Z77
# Kernel: 4.15.3-2-ARCH x86_64
# Processor: Intel(R) Core(TM) i7-3770 CPU @ 3.40GHz (6/58/9)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Module cpuid loaded successfully.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Intel digital thermal sensor...                             Success!
    (driver `coretemp')
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found `Nuvoton NCT5532D/NCT6779D Super IO Sensors'          Success!
    (address 0x290, driver `nct6775')
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no): 
Probing for `IPMI BMC KCS' at 0xca0...                      No
Probing for `IPMI BMC SMIC' at 0xca8...                     No

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (yes/NO): 

Lastly, we can probe the I2C/SMBus adapters for connected hardware
[... 36 lines cut ...]

Next adapter: NVIDIA i2c adapter 6 at 1:00.0 (i2c-3)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: NVIDIA i2c adapter 8 at 1:00.0 (i2c-4)
Do you want to scan it? (yes/NO/selectively): 

Now follows a summary of the probes I have just done.
Just press ENTER to continue: 

Driver `nct6775':
  * ISA bus, address 0x290
    Chip `Nuvoton NCT5532D/NCT6779D Super IO Sensors' (confidence: 9)

Driver `coretemp':
  * Chip `Intel digital thermal sensor' (confidence: 9)

Do you want to overwrite /etc/conf.d/lm_sensors? (YES/no): 
Unloading i2c-dev... OK
Unloading cpuid... OK
```

> You can use sensors command to check it.

It's almost empty before the last question about conf overwritting. Do you suggest me to trying regenerate config with different choices unless I'll get values without zeros?

UshakovVasilii, 2018-02-22:
you also can try to execute `# yes | sensors-detect`.

To ignore sensor you need to create file with chip and ignore lines, sample for DELL sensor:
File `/etc/sensors.d/dell`:
```
chip "dell_smm-virtual-0"
 ignore fan1
```

AlexWayfer, 2018-02-22:
> you also can try to execute # yes | sensors-detect.

The same result, as `auto`.

> To ignore sensor you need to create file with chip and ignore lines

For my output I'm trying file `/etc/sensors.d/disabling`:

```
chip "nct6779-isa-0290"
 ignore AUXTIN0
```

And it doesn't help :(

UshakovVasilii, 2018-02-27:
I think `AUXTIN0` is label. You can check label list in `/etc/sensors3.conf`.

I'm sorry for late reply.

AlexWayfer, 2018-02-28:
> I think AUXTIN0 is label. You can check label list in /etc/sensors3.conf.

There is what I found by `aux`:

```
chip "fscpos-*" "fscher-*"
#FSC "Hermes"

    label in0 "+12V"
    label in1 "+5V"
    label in2 "Vbat"

    label temp1 "CPU Temp"
    label temp2 "M/B Temp"
    label temp3 "Aux Temp"

chip "fscscy-*"
#FSC "Scylla"

    label in0 "+12V"
    label in1 "+5V"
    label in2 "+3.3V"

    label temp1 "CPU0 Temp"
    label temp2 "CPU1 Temp"
    label temp3 "M/B Temp"
    label temp4 "Aux Temp"
```

And what can I do with this?

> I'm sorry for late reply.

No problem, thanks for help.

AlexWayfer, 2018-02-28:
> Looks like file should be like:
> ```
> chip "nct6779-isa-0290"
>  ignore temp1
>  ignore temp2
>  ignore temp3
>  ignore temp4
> ```

Wow! It disabled `AUXTIN0` and `1`, but `2` and `3` still here.

There is content of my `/etc/sensors3.conf`:

<details>
  <summary>Content</summary>
  <p>

```
# libsensors configuration file
# -----------------------------
#
# This default configuration file only includes statements which do not
# differ from one mainboard to the next. Only label, compute and set
# statements for internal voltage and temperature sensors are included.
#
# In general, local changes should not be added to this file, but rather
# placed in custom configuration files located in /etc/sensors.d. This
# approach makes further updates much easier.
#
# Such custom configuration files for specific mainboards can be found at
# http://www.lm-sensors.org/wiki/Configurations

chip "lm78-*" "lm79-*" "lm80-*" "lm96080-*"

    label temp1 "M/B Temp"

chip "w83792d-*"

    label in0 "VcoreA"
    label in1 "VcoreB"
    label in6 "+5V"
    label in7 "5VSB"
    label in8 "Vbat"

    set in6_min  5.0 * 0.90
    set in6_max  5.0 * 1.10
    set in7_min  5.0 * 0.90
    set in7_max  5.0 * 1.10
    set in8_min  3.0 * 0.90
    set in8_max  3.0 * 1.10

chip "w83793-*"

    label in0 "VcoreA"
    label in1 "VcoreB"
    label in7 "+5V"
    label in8 "5VSB"
    label in9 "Vbat"

    set in7_min  5.0 * 0.90
    set in7_max  5.0 * 1.10
    set in8_min  5.0 * 0.90
    set in8_max  5.0 * 1.10
    set in9_min  3.0 * 0.90
    set in9_max  3.0 * 1.10

chip "w83795g-*" "w83795adg-*"

    label in12 "+3.3V"
    label in13 "3VSB"
[... 414 lines cut ...]
    set in2_min  3.3 * 0.90
    set in2_max  3.3 * 1.10

    label temp2 "Board Temp"

chip "adt7476-*" "adt7490-*"

    label in1 "Vcore"
    label in2 "+3.3V"
    label in3 "+5V"
    label in4 "+12V"

    set in2_min  3.3 * 0.90
    set in2_max  3.3 * 1.10
    set in3_min  5.0 * 0.90
    set in3_max  5.0 * 1.10
# Depending on how your ADT7476 is hardwired, you may or may not have
# +12V readings.
#    set in4_min 12.0 * 0.90
#    set in4_max 12.0 * 1.10

    label temp2 "M/B Temp"
```
  </p>
</details>

UshakovVasilii, 2018-02-28:
> Wow! It disabled AUXTIN0 and 1, but 2 and 3 still here.

Possible AUXTIN0 is temp4, so you can try:
```
chip "nct6779-isa-0290"
 ignore temp4
 ignore temp5
 ignore temp6
 ignore temp7
```

AlexWayfer, 2018-02-28:
> Possible AUXTIN0 is temp4, so you can try:

No, `AUXTIN0` is `temp3` (`AUXTIN1` was disabled too, it's `temp4`), so `temp[3-6]` helps. Thank you.

OK, I understood the principe. :joy: 

`temp2` — `CPUTIN`, `8` — `PCH_CHIP_CPU_MAX_TEMP`, `9` — `PCH_CHIP_TEMP` and `10` — `PCH_CPU_TEMP`.

Now there is no sensors with zero tempatures.

Thank you!

Sorry for unrelated to extension questions.

I just thought that you can add an option to extension like `Hide zero values`.

closed 2018-02-28 by AlexWayfer