UshakovVasilii/gnome-shell-extension-freon #86: Cannot remove configuration

lavanoid, 2018-03-12:
I'm having an issue where I can no longer open the Freon menu. I can see my CPU temperatures at the bottom of the screen but I cannot make any changes as the menu will refuse to open when I click on it.

I'm currently trying to find the configuration file, so I can delete it. Uninstalling Freon does not resolve the issue as it saves the configuration file somewhere, so it remembers settings when I reinstall it :(

UshakovVasilii, 2018-03-13:
Hi, I did not understand what menu you are talking about.
You can open extension configuration from `gnome-tweak-tool`. Or `dconf-editor` config is in `/org/gnome/shell/extensions/freon`

lavanoid, 2018-05-24:
Thanks for the response. The issue I was experiencing was with the Freon icon that would be in the notification area. It would show some temperatures of devices I had previously selected but every time I clicked on it to open the menu (so I could select/deselect devices to monitor the temperatures of), the menu wouldn't open at all x_x

UshakovVasilii, 2018-05-25:
Freon icon should be on top panel, I don't understand what icon do on notification area. Do you have any error in journalctl log?

lavanoid, 2018-05-27:
When I click the icon, it does nothing when normally a small menu should appear where I can select which device temperatures to monitor. I'm not sure how much clearer I can put this across. I'll have to do some more testing and then I'll provide logs/images of what is happening.

UshakovVasilii, 2019-04-04:
deprecated

closed 2019-04-04 by UshakovVasilii