UshakovVasilii/gnome-shell-extension-freon #89: Mention lm-sensors requirement in README/GNOME Page

fgblomqvist, 2018-06-25:
It is afaik not mentioned anywhere that one has to install lm-sensors to get this to work. Should be added to the readme and the gnome extensions page.

UshakovVasilii, 2018-06-27:
Hello, we have note in https://github.com/UshakovVasilii/gnome-shell-extension-freon/wiki
and https://github.com/UshakovVasilii/gnome-shell-extension-freon/wiki/Dependency page.
Also if lm-sensors not installed you should see message after click on freon icon.

fgblomqvist, 2018-06-27:
Okay. The message didn't work. Clicking on the freon icon did nothing. I still think that it should be in the README/Gnome page since it's critical for it to work (e.g. it's not a feature) but I suppose it's fine as long as the message works.

commit d5b37692 (2019-04-04): desplay menu if lm_sensors is not intstalled. fix #89
closed 2019-04-04 by UshakovVasilii