UshakovVasilii/gnome-shell-extension-freon #98: Crashes at launch on Gnome 3.30

Vash63, 2018-09-09:
I'm seeing the following in my logs and the extension gets marked as errored in Tweaks.

```sep 09 11:18:41 vashnix gnome-shell[1067]: Extension "freon@UshakovVasilii_Github.yahoo.com" had error: SyntaxError: missing ( after for
sep 09 11:18:41 vashnix gnome-shell[1067]: Some code called array.toString() on a Uint8Array instance. Previously this would have interpreted the bytes of the array as a string, but that is nonstandard. In the future this will return the bytes as comma-separated digits. For the time being, the old behavior has been preserved, but please fix your code anyway to explicitly call ByteArray.toString(array).
sep 09 11:18:41 vashnix gnome-shell[1067]: Some code called array.toString() on a Uint8Array instance. Previously this would have interpreted the bytes of the array as a string, but that is nonstandard. In the future this will return the bytes as comma-separated digits. For the time being, the old behavior has been preserved, but please fix your code anyway to explicitly call ByteArray.toString(array).
sep 09 11:18:41 vashnix gnome-shell[1067]: Some code called array.toString() on a Uint8Array instance. Previously this would have interpreted the bytes of the array as a string, but that is nonstandard. In the future this will return the bytes as comma-separated digits. For the time being, the old behavior has been preserved, but please fix your code anyway to explicitly call ByteArray.toString(array).
sep 09 11:18:41 vashnix gnome-shell[1067]: Some code called array.toString() on a Uint8Array instance. Previously this would have interpreted the bytes of the array as a string, but that is nonstandard. In the future this will return the bytes as comma-separated digits. For the time being, the old behavior has been preserved, but please fix your code anyway to explicitly call ByteArray.toString(array).```

UshakovVasilii, 2018-09-09:
I'll fix after Gnome 3.30 come to ArchLinux

AlexWayfer, 2018-09-09:
> I'll fix after Gnome 3.30 come to ArchLinux

Just FYI, you can grab it from `testing` repo: https://www.archlinux.org/packages/testing/x86_64/gnome-shell/

Vash63, 2018-09-09:
Thanks. I'm on 3.30 on arch's testing, I'll reinstall Freon once it's updated.

tingvarsson, 2018-09-17:
fyi if you havent noticed 3.30 is now on Arch non-testing. :)

yaroslav-shabalin, 2018-09-18:
> I'll fix after Gnome 3.30 come to ArchLinux
The fix is very much awaited.

UshakovVasilii, 2018-09-18:
I'm fixing...