UshakovVasilii/gnome-shell-extension-freon #100: "[FREON] Render all MenuItems" spammed twice at extension start

tingvarsson, 2018-09-20:
While (re-)starting Gnome Shell there are some logs from freon with little value to the syslog:

> sep 20 11:43:24 umaro gnome-shell[1022]: **[FREON] Render all MenuItems**
> sep 20 11:43:25 umaro gnome-shell[1022]: **[FREON] Render all MenuItems**
> sep 20 11:43:25 umaro gnome-shell[1022]: GNOME Shell started at Thu Sep 20 2018 11:43:23 GMT+0200 (CEST)

Saw in the code that other log statements are commented out on master, should this be commented out as well?

UshakovVasilii, 2018-09-21:
I keep this line for detect problems (Full rerander take a lot of resources). At start it is OK to render all MenuItems twice, for example we can get information about sensors after information about disks (We cannot get information about all sensors in same time).

tingvarsson, 2018-09-21:
I understand, was hoping there was some better logging framework but it seems there isn't. e.g. to have debug logging only when enabled or so. Have to invent something on the extension side, and then it is possibly not worth it.
Easiest would probably be to have a setting on/off debug mode somewhere (possible to hide in a gsetting or similar?) and create a global debug function that checks if set or not and calls log based on that.
Can play around with it if you would be OK with a PR on that? (totally OK if not)

UshakovVasilii, 2018-09-21:
Yeah, we can add option. But it will make the application a bit more complicated. I want to keep application simple as it possible. I created #103

closed 2018-09-21 by UshakovVasilii