UshakovVasilii/gnome-shell-extension-freon #106: Global graphic freezes

AlexWayfer, 2018-10-12:
Hello!

I had periodic freezes in my GNOME, and it was noticeable while I watched streams in YouTube, for example.

Now I faced them again, in Google Play Music. There is no sound lags, only graphic system-wide freeze for 50–200 ms. Every animations stops. And I noticed that it occurs with the same interval. I decided to figure out what's causing this. I began turn off GNOME extensions one by one, and here I'm: Freon freezes my system.

Settings window:

[image]

And yeah, it occurs every 5 seconds.

Disabling Hddtemp reduce freeze time, disabling NVIDIA reduce to almost zero. But GPU temp is important for me.

About my system: Intel i5 6600, GTX 1060 6 GB, Arch Linux, GNOME `3.30.1`.

I'm ready to help: measures, tests, etc. But I think this way of working is wrong for extension. Maybe, we should poll in the separated thread. Maybe, we can simplify polling. Maybe, there is no way to improve this situation.

UshakovVasilii, 2018-10-12:
Hello!
You can disable extension and check commands for monitoring.
Extension use `nvidia-settings -q gpucoretemp -t` command for NVIDIA card temperature.
Also you can check `sesnsors` command.
Also Udisk2 better than Hddtemp.

If you have lag after `sesnsors` command - https://github.com/UshakovVasilii/gnome-shell-extension-freon/wiki/FAQ#system-to-hang-every-n-second

If you have lag after `nvidia-settings` command than we need to find another approach, also problem can be related to NVIDIA software.

AlexWayfer, 2018-10-12:
```
time sensors

...

real	0m0.021s
user	0m0.006s
sys	0m0.015s
```

```
time nvidia-settings -q gpucoretemp -t

...

real	0m0.174s
user	0m0.072s
sys	0m0.046s
```

`sensors` looks fast, but `nvidia` has 150+ ms time. As I wrote above, the whole annoying lag is summed up from `Hddtemp` and `Nvidia`.

UshakovVasilii, 2018-10-12:
Do you have video lag after `nvidia-settings -q gpucoretemp -t`?

AlexWayfer, 2018-10-12:
> Do you have video lag after nvidia-settings -q gpucoretemp -t?

No, even after 10 times there is no lag:

```
$ time for i in {1..10}; do nvidia-settings -q gpucoretemp -t; done

...

real	0m1.074s
user	0m0.458s
sys	0m0.346s
```

UshakovVasilii, 2018-10-12:
ok, maybe you have lag after `sensors` command?

Also you can check hddtemp `nc localhost <PORT>`.
To detect port:
```
systemctl show hddtemp.service -p MainPID
```
or
```
pidof hddtemp
/proc/<PID>/cmdline
```
default 7634

AlexWayfer, 2018-10-12:
> ok, maybe you have lag after sensors command?

No, `sensors` is faster and doesn't cause lags:

```
time for i in {1..10}; do sensors; done

...

real	0m0.067s
user	0m0.033s
sys	0m0.033s
```

> Also you can check hddtemp nc localhost <PORT>.

`pidof hddtemp` and `systemctl show hddtemp.service -p MainPID` returns `474`, but `nc localhost 474` returns (output) nothing.

UshakovVasilii, 2018-10-12:
Hm,
Is `systemctl show hddtemp.service -p ActiveState` command output has `ActiveState=active` (should be second line)

AlexWayfer, 2018-10-12:
> Is systemctl show hddtemp.service -p ActiveState command output has ActiveState=active (should be second line)

Yes.

> `/proc/<PID>/cmdline`

I don't know why you're writing this line, what should I do with it. As I wrote above, `pidof hddtemp` returns `474`.

UshakovVasilii, 2018-10-12:
`show hddtemp.service -p ActiveState` has `ActiveState=active`.
port is 474.
but `nc localhost 474` output nothing.
It is strange, is extension show disk temperature?

UshakovVasilii, 2018-10-12:
maybe you have some strange logs in `# journalctl -f`

UshakovVasilii, 2018-10-12:
And what extension version do you use? From extension.gnome.org or git or AUR?

AlexWayfer, 2018-10-12:
> It is strange, is extension show disk temperature?

Yeah, for SSD. But I also have 2 (mounted) HDDs.

> maybe you have some strange logs in # journalctl -f?

I don't know how to check if some log is strange. :sweat_smile: 

> And what extension version do you use? From extension.gnome.org or git or AUR?

extension.gnome.org, the latest version.

AlexWayfer, 2018-10-12:
Here are logs from `journalctl -f` (almost everything from Freon, I guess):

```
-- Logs begin at Mon 2018-07-23 18:22:32 MSK. --
Oct 12 20:30:20 Alex-ArchMainPC gnome-shell[1018]: Some code called array.toString() on a Uint8Array instance. Previously this would have interpreted the bytes of the array as a string, but that is nonstandard. In the future this will return the bytes as comma-separated digits. For the time being, the old behavior has been preserved, but please fix your code anyway to explicitly call ByteArray.toString(array).
                                                   (Note that array.toString() may have been called implicitly.)
                                                   0 _init/childWatch<() ["/home/alex/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/nvidiaUtil.js":36]
                                                   1 () ["self-hosted":979]
Oct 12 20:30:20 Alex-ArchMainPC gnome-shell[1018]: Some code called array.toString() on a Uint8Array instance. Previously this would have interpreted the bytes of the array as a string, but that is nonstandard. In the future this will return the bytes as comma-separated digits. For the time being, the old behavior has been preserved, but please fix your code anyway to explicitly call ByteArray.toString(array).
                                                   (Note that array.toString() may have been called implicitly.)
                                                   0 execute/childWatch<() ["/home/alex/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/commandLineUtil.js":36]
                                                   1 () ["self-hosted":979]
Oct 12 20:30:21 Alex-ArchMainPC gnome-shell[1018]: [FREON] Render all MenuItems
Oct 12 20:30:38 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 1008ff11 with keysym 1008ff11 (keycode 7a).
Oct 12 20:30:38 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 1008ff12 with keysym 1008ff12 (keycode 79).
Oct 12 20:30:38 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 33 with keysym 33 (keycode c).
Oct 12 20:30:38 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 1008ff13 with keysym 1008ff13 (keycode 7b).
Oct 12 20:30:38 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 31 with keysym 31 (keycode a).
Oct 12 20:30:38 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 32 with keysym 32 (keycode b).
Oct 12 20:30:38 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 34 with keysym 34 (keycode d).
```

AlexWayfer, 2018-10-12:
And there are now many-many `Window manager warning`, also one `ERROR:page_load_metrics_update_dispatcher.cc` (I don't know, is it from Freon):

```
Oct 12 20:30:38 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 1008ff11 with keysym 1008ff11 (keycode 7a).
Oct 12 20:30:38 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 1008ff12 with keysym 1008ff12 (keycode 79).
Oct 12 20:30:38 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 33 with keysym 33 (keycode c).
Oct 12 20:30:38 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 1008ff13 with keysym 1008ff13 (keycode 7b).
Oct 12 20:30:38 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 31 with keysym 31 (keycode a).
Oct 12 20:30:38 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 32 with keysym 32 (keycode b).
Oct 12 20:30:38 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 34 with keysym 34 (keycode d).
Oct 12 20:33:49 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: [2603:2603:1012/203349.449677:ERROR:page_load_metrics_update_dispatcher.cc(166)] Invalid first_paint 3.354 s for first_text_paint 3.17 s
Oct 12 20:34:29 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 1008ff11 with keysym 1008ff11 (keycode 7a).
Oct 12 20:34:29 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 1008ff12 with keysym 1008ff12 (keycode 79).
Oct 12 20:34:29 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 33 with keysym 33 (keycode c).
Oct 12 20:34:29 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 1008ff13 with keysym 1008ff13 (keycode 7b).
Oct 12 20:34:29 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 31 with keysym 31 (keycode a).
Oct 12 20:34:29 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 32 with keysym 32 (keycode b).
Oct 12 20:34:29 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 34 with keysym 34 (keycode d).
Oct 12 20:34:46 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 1008ff11 with keysym 1008ff11 (keycode 7a).
Oct 12 20:34:46 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 1008ff12 with keysym 1008ff12 (keycode 79).
Oct 12 20:34:46 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 33 with keysym 33 (keycode c).
Oct 12 20:34:46 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 1008ff13 with keysym 1008ff13 (keycode 7b).
Oct 12 20:34:46 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 31 with keysym 31 (keycode a).
Oct 12 20:34:46 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 32 with keysym 32 (keycode b).
Oct 12 20:34:46 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 34 with keysym 34 (keycode d).
Oct 12 20:34:59 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 1008ff11 with keysym 1008ff11 (keycode 7a).
Oct 12 20:34:59 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 1008ff12 with keysym 1008ff12 (keycode 79).
Oct 12 20:34:59 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 33 with keysym 33 (keycode c).
Oct 12 20:34:59 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 1008ff13 with keysym 1008ff13 (keycode 7b).
Oct 12 20:34:59 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 31 with keysym 31 (keycode a).
Oct 12 20:34:59 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 32 with keysym 32 (keycode b).
Oct 12 20:34:59 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 34 with keysym 34 (keycode d).
Oct 12 20:35:12 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 1008ff11 with keysym 1008ff11 (keycode 7a).
Oct 12 20:35:12 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 1008ff12 with keysym 1008ff12 (keycode 79).
Oct 12 20:35:12 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 33 with keysym 33 (keycode c).
[... cut ...]
Oct 12 20:35:43 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 1008ff13 with keysym 1008ff13 (keycode 7b).
Oct 12 20:35:43 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 31 with keysym 31 (keycode a).
Oct 12 20:35:43 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 32 with keysym 32 (keycode b).
Oct 12 20:35:43 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 34 with keysym 34 (keycode d).
```

AlexWayfer, 2018-10-12:
OK, there is not one `ERROR:page_load_metrics_update_dispatcher.cc`:

```
Oct 12 20:36:39 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 32 with keysym 32 (keycode b).
Oct 12 20:36:39 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 34 with keysym 34 (keycode d).
Oct 12 20:36:42 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: [2603:2603:1012/203642.773524:ERROR:page_load_metrics_update_dispatcher.cc(190)] Invalid first_paint 1.42 s for first_meaningful_paint 1.256 s
Oct 12 20:36:45 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: [2603:2603:1012/203645.197118:ERROR:page_load_metrics_update_dispatcher.cc(190)] Invalid first_paint 1.42 s for first_meaningful_paint 1.256 s
Oct 12 20:36:47 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: [2603:2603:1012/203647.755424:ERROR:page_load_metrics_update_dispatcher.cc(166)] Invalid first_paint 3.354 s for first_text_paint 3.17 s
Oct 12 20:36:50 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: [2603:2603:1012/203650.224836:ERROR:page_load_metrics_update_dispatcher.cc(190)] Invalid first_paint 1.42 s for first_meaningful_paint 1.256 s
Oct 12 20:36:51 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: [2603:2603:1012/203651.256608:ERROR:page_load_metrics_update_dispatcher.cc(190)] Invalid first_paint 1.42 s for first_meaningful_paint 1.256 s
Oct 12 20:36:52 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: [2603:2603:1012/203652.743103:ERROR:page_load_metrics_update_dispatcher.cc(190)] Invalid first_paint 1.42 s for first_meaningful_paint 1.256 s
Oct 12 20:36:58 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: [2603:2603:1012/203658.397553:ERROR:page_load_metrics_update_dispatcher.cc(190)] Invalid first_paint 1.42 s for first_meaningful_paint 1.256 s
Oct 12 20:37:00 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: [2603:2603:1012/203700.258231:ERROR:page_load_metrics_update_dispatcher.cc(190)] Invalid first_paint 1.42 s for first_meaningful_paint 1.256 s
Oct 12 20:37:09 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: [2603:2603:1012/203709.298073:ERROR:page_load_metrics_update_dispatcher.cc(190)] Invalid first_paint 1.42 s for first_meaningful_paint 1.256 s
Oct 12 20:37:16 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: [2603:2603:1012/203716.393326:ERROR:page_load_metrics_update_dispatcher.cc(190)] Invalid first_paint 1.42 s for first_meaningful_paint 1.256 s
Oct 12 20:37:23 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: [2603:2603:1012/203723.310622:ERROR:page_load_metrics_update_dispatcher.cc(190)] Invalid first_paint 1.42 s for first_meaningful_paint 1.256 s
Oct 12 20:37:38 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: [2603:2603:1012/203738.591841:ERROR:page_load_metrics_update_dispatcher.cc(190)] Invalid first_paint 1.42 s for first_meaningful_paint 1.256 s
Oct 12 20:37:41 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: [2603:2603:1012/203741.859639:ERROR:page_load_metrics_update_dispatcher.cc(190)] Invalid first_paint 1.42 s for first_meaningful_paint 1.256 s
Oct 12 20:37:42 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: [2603:2603:1012/203742.884927:ERROR:page_load_metrics_update_dispatcher.cc(190)] Invalid first_paint 1.42 s for first_meaningful_paint 1.256 s
Oct 12 20:37:45 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: [2603:2603:1012/203745.462787:ERROR:page_load_metrics_update_dispatcher.cc(190)] Invalid first_paint 1.42 s for first_meaningful_paint 1.256 s
Oct 12 20:37:47 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 1008ff11 with keysym 1008ff11 (keycode 7a).
Oct 12 20:37:47 Alex-ArchMainPC org.gnome.Shell.desktop[1018]: Window manager warning: Overwriting existing binding of keysym 1008ff12 with keysym 1008ff12 (keycode 79).
```

UshakovVasilii, 2018-10-12:
Ah, I'm sorry, I missed `return` in code))
Please check `hddtemp` command with video/audio

AlexWayfer, 2018-10-12:
> Please check hddtemp command with video/audio

Sorry, I didn't understand. `hddtemp` has output `Too few arguments: you must specify one drive, at least.`.

UshakovVasilii, 2018-10-12:
Got it. I have not idea for now, I will check on computer with NVIDIA...

AlexWayfer, 2018-10-12:
Thank you. Maybe extra logs are causing these freezes… Idk too.

UshakovVasilii, 2018-10-12:
You also can check version from git https://github.com/UshakovVasilii/gnome-shell-extension-freon#installation-from-git

UshakovVasilii, 2018-10-13:
I checked extension with NVIDIA card and hddtemp, I have no freezes. Also I have no extra logs.

UshakovVasilii, 2018-10-13:
>> It is strange, is extension show disk temperature?

> Yeah, for SSD. But I also have 2 (mounted) HDDs.

You can configure hddtemp  for HDDs https://wiki.archlinux.org/index.php/Hddtemp "Override default disk"

But better to use [udisks2](https://wiki.archlinux.org/index.php/udisks), _SMART can be enabled in BIOS_

UshakovVasilii, 2018-10-13:
Ah, possible problem related to browser (GPU acceleration, Flash or so), please check in different browsers)))

AlexWayfer, 2018-10-16:
> Ah, possible problem related to browser (GPU acceleration, Flash or so), please check in different browsers)))

The same in Firefox.

tingvarsson, 2018-10-18:
I noticed the same issue today. I noticed it by dragging a window around a lot, and saw that the window glitched everytime freon updated its values.
Also tested with another nvidia extension that queries nvidia-settings periodically. Same glitches experienced there.

> > Do you have video lag after nvidia-settings -q gpucoretemp -t?
> 
> No, even after 10 times there is no lag:
> 
> ```
> $ time for i in {1..10}; do nvidia-settings -q gpucoretemp -t; done
> 
> ...
> 
> real	0m1.074s
> user	0m0.458s
> sys	0m0.346s
> ```
> Maybe it's something extensions-specific.

I think these numbers actually show the issue. Have to remember that extensions are run in the gnome-shell thread, and not as their own. E.g. it probably blocks gnome-shell for a while! And it is gnome-shell/mutter that most likely gives the small glitches.

tingvarsson, 2018-10-18:
Played around with nvidia-smi and it seems quicker to read out the temp:
```
(! 5755)-> time for i in {1..10}; do nvidia-smi --query-gpu=temperature.gpu --format=csv,noheader; done
55
55
55
55
55
55
55
55
55
55

real	0m0,074s
user	0m0,011s
sys	0m0,017s
```

And my time on nvidia-settings for reference:
```
(! 5756)-> time for i in {1..10}; do nvidia-settings -q gpucoretemp -t; done
53
53
53
53
53
53
53
53
53
53
53
53
53
53
53
53
53
53
53
53

real	0m0,580s
user	0m0,460s
sys	0m0,090s

```

Also has the benefit of not giving duplicated values. Not sure how it will look in a multi GPU setup however.

AlexWayfer, 2019-01-24:
> But better to use udisks2

I have 99°C somewhy…

```
> udisksctl dump | grep -i temp
    SmartTemperature:                           372.15000000000003
    SmartTemperature:                           308.15000000000003
    SmartTemperature:                           297.15000000000003
```

EUA, 2020-01-10:
Yeah,
For example, if a hdd delays "smartctl" command, than freon waits it's answer and make delay gnome-shell leads micro-freezes.

I don't know/like JS. Don't know how to implement but...  It can be solved via **async** sensor reading.

NexonSU, 2020-04-25:
Well, @UshakovVasilii will you fix that sometime? :D
I don't want to turn it off.

jonasmalacofilho, 2020-04-26:
Most of this thread relates to delays caused by nvidia-settings.  That should have already been solved by #162 (" Switch to nvidia-smi to reduce CPU load when fetching GPU temperatures").

---

For those already on Linux 5.6, check out the new drivetemp driver and disable any custom HDD temperature provider in Freon.

Another option is switching Freon to use udisks (AFAIK it doesn't wake up suspended disks, though that can sometimes cause outdated temperatures to be reported).

mvrk69, 2024-09-13:
I had this issue with freon on Fedora 37 (Xorg) and culprit was smartctl, after disabling smartclt and use udisks2 instead everything was good. 
But now i'm on Fedora 40 (Xorg) and the problem returned. I usually had the lmsensors, nvidia and udisks2 enabled. 
I did some tests and i get stutter with just lmsensors, also with just nvidia, if i leave just udisks2 doesn't cause stutter, but i loose temp for cpu and gpu.
So basically i have to disable the extension everytime i want to play a video or a game.