UshakovVasilii/gnome-shell-extension-freon #107: List with NaN when NVIDIA is disabled

yustnip, 2018-10-18:
After enabled Intel PRIME Profile via nvidia-prime I got the NVIDIA-list with NaN values.

[image]

When NVIDIA PRIME Profile is enabled all's fine.

[image]

Video Card Temperature Utility is NVIDIA.

UshakovVasilii, 2018-10-19:
Please provide `nvidia-settings -q gpucoretemp -t` and `nvidia-settings -q gpus` output for first case.

yustnip, 2018-10-20:
nvidia-settings -q gpucoretemp -t

```

ERROR: NVIDIA driver is not loaded

ERROR: Error querying enabled displays on GPU 0 (Missing Extension).

ERROR: Error querying connected displays on GPU 0 (Missing Extension).

ERROR: Error resolving target specification '' (No targets match target
       specification), specified in query 'gpucoretemp'.

```

nvidia-settings -q gpus

```

ERROR: NVIDIA driver is not loaded

ERROR: Error querying enabled displays on GPU 0 (Missing Extension).

ERROR: Error querying connected displays on GPU 0 (Missing Extension).

```

yustnip, 2018-10-20:
Possible solution - https://github.com/UshakovVasilii/gnome-shell-extension-freon/pull/108

UshakovVasilii, 2018-10-21:
Merged

closed 2018-10-21 by UshakovVasilii