UshakovVasilii/gnome-shell-extension-freon #109: Can't find anything but SSD/HDD stats

chrysophylax, 2018-10-29:
Hi, I installed the Extension straight from git ab13ab47548199b5574996dbc873c16b8b5888ee
and I can't seem to get *anything* but disk temperature stats.  I have an Nvidia Quadro card and have nvidia-settings installed and working.

OS: RHEL 7.5 Maipo
Program Version: [ab13ab](https://github.com/UshakovVasilii/gnome-shell-extension-freon/commit/ab13ab47548199b5574996dbc873c16b8b5888ee)
GNOME Version: 3.25.4
lm_sensors: version 3.4.0 with libsensors version 3.4.0
CPU: Intel(R) Xeon(R) E-2176M  CPU @ 2.70GHz

1. Running `sensors` gives me the following

```
iwlwifi-virtual-0
Adapter: Virtual device
temp1:        +44.0°C  

thinkpad-isa-0000
Adapter: ISA adapter
fan1:           0 RPM

acpitz-virtual-0
Adapter: Virtual device
temp1:        +65.0°C  (crit = +128.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +76.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +62.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +68.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +64.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +64.0°C  (high = +100.0°C, crit = +100.0°C)
Core 4:        +62.0°C  (high = +100.0°C, crit = +100.0°C)
Core 5:        +76.0°C  (high = +100.0°C, crit = +100.0°C)

pch_cannonlake-virtual-0
Adapter: Virtual device
temp1:        +56.0°C  
```

2. journalctl -xr after install has the following infinite lines repeating
```
Oct 29 18:13:19 iris gnome-shell[2938]: SyntaxError: JSON.parse: unexpected character at line 1 column 2 of the JSON data
Oct 29 18:13:19 iris gnome-shell[2938]: SyntaxError: JSON.parse: unexpected character at line 1 column 2 of the JSON data
Oct 29 18:13:19 iris gnome-shell[2938]: SyntaxError: JSON.parse: unexpected character at line 1 column 2 of the JSON data
```

Any suggestions? Thanks.

UshakovVasilii, 2018-10-29:
Hello, please provide `sensors -j` output

UshakovVasilii, 2018-10-30:
Got it, please provide `sensors --version`.

UshakovVasilii, 2018-10-30:
Sensors feature https://github.com/lm-sensors/lm-sensors/pull/21

AlexWayfer, 2018-10-30:
> Hm, last release was 3.4.0 - Jun 25, 2015
> But -j option was added Apr 14, 2017

I think you can ask a new release.

UshakovVasilii, 2018-11-01:
Information about new release https://github.com/lm-sensors/lm-sensors/issues/29#issuecomment-434383258

UshakovVasilii, 2018-11-01:
related https://github.com/lm-sensors/lm-sensors/issues/29

rodgersan, 2018-11-04:
Hello,

I am facing more or less the same situation on Fedora 29 with the lm_sensors-3.4.0-17.20180522git70f7e08.fc29.x86_64 package and Freon master version.

"sensors -j" outputs the following error for my wifi (when antenna is off):
`{
   "iwlwifi-virtual-0":{
      "Adapter": "Virtual device",
      "Wifi temp":{
ERROR: Can't get value of subfeature temp1_input: Can't read
      }`

Whereas "sensors" just outputs a N/A.

Everything is fine when wifi antenna is on.

I guess this is more a lm_sensors issue yet it would be nice if you could do the same for ERROR as for N/A.

Regards,

russel, 2018-11-13:
I believe I have this problem on Debian Sid, fully up to date, using Freon from the Git repository up to date.
[image]

/usr/bin/sensors has no -j option.

```
|> dpkg -l lm-sensors
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-=================================================
ii  lm-sensors     1:3.4.0-4    amd64        utilities to read temperature/voltage/fan sensors
```

UshakovVasilii, 2018-11-13:
Hi all, you can use extension from https://extensions.gnome.org/extension/841/freon/ or 986ddf8d962f9f661641b216fd4ef5933bdfb903 version for now.

I think need to roll back `-j` option. World not ready for that.. )

russel, 2018-11-14:
It seems that Fedora Rawhide is ready for the -j option:
```
[root@anglides ~]# dnf list lm_sensors
Last metadata expiration check: 0:14:28 ago on Wed 14 Nov 2018 07:20:37 GMT.
Installed Packages
lm_sensors.x86_64               3.4.0-18.20180522git70f7e08.fc30               @rawhide
```
but as Debian Sid seems not to be, and it is what I use most, I shall go with the -j-less option. :-)

chrysophylax, 2018-11-28:
Thanks! Looking forward to try out version 986ddf8

UshakovVasilii, 2018-11-29:
November 23rd, 2018: lm-sensors 3.5.0 Released! Previous release was in 2015

russel, 2018-12-22:
After some struggles in Debian Experimental, lm-sensors package in Debian Sid is now 3.5.0 and all the sensors are back in the Freon display and are working as they should.

Thanks @UshakovVasilii  for Freon.

brittyazel, 2019-03-22:
I too am having this issue, though I only have it on my Ryzen system, on my Intel system everything is fine. I'm on Archlinux with GNOME 3.32 and lm-sensors 3.5.0

UshakovVasilii, 2019-05-24:
We need lm_sensors 3.5.0+

Wrong JSON output fixed by #114 #66 #133

closed 2019-05-24 by UshakovVasilii