UshakovVasilii/gnome-shell-extension-freon #110: Negative voltages has a double minus signs.

squall77it, 2018-10-30:
Negative voltages has a double minus signs, Ex: -12.55V is --12.55V.

I fix it modifing the line 377 of extensions.js from:
`value:_("%s%.2fV").format(((voltage.volt >= 0) ? '+' : '-'),`
to
`value:_("%s%.2fV").format(((voltage.volt >= 0) ? '+' : ''),`
because numbers less then 0 already have the minus sign.

Before:
[image]
After:
[image]

AlexWayfer, 2018-10-31:
Negative voltage — is it OK?

AlexWayfer, 2018-10-31:
> Yes is OK
> https://en.wikipedia.org/wiki/ATX#Power_supply

Thank you!

commit 9ecff762 (2018-11-01): fix double voltages minus, resolve #110
closed 2018-11-01 by UshakovVasilii