UshakovVasilii/gnome-shell-extension-freon #111: Extension fails on even harmless 'sensors' errors

internetstaff, 2018-11-04:
Since CommandLineUtil appends stderr output to the JSON output of sensors -A -j, the extension fails to parse the JSON any time there's an error from sensors. 

Removing stderr from 'output' allows the extension to work despite harmless errors from sensors.

It's not obvious to me that the extension makes use of any stderr output, but at least in the case of sensors, it seems like CommandLineUtil could make appending stderr optional.

It could perhaps alternatively display the error - but that seems like overkill. Right now the extension just says "run sensors-detect" which in many cases won't fix a benign sensor problem.

arigit, 2019-05-12:
see a workoround for this in #114 , via script. 
lm-sensors sadly produces non-compliant JSON when it runs into errors, and the script solves that.
I hope the woraround can be incorporated into freon to save people the hassle of dealing with this

UshakovVasilii, 2019-05-17:
fixed by #114

closed 2019-05-17 by UshakovVasilii

arigit, 2019-05-17:
Tested just in case. Works great
Thanks!