UshakovVasilii/gnome-shell-extension-freon #114: Clarify notes and Catalyst dependencies

terrycloth, 2018-11-17:
Currently, the wiki says displaying AMD "Catalyst" temperatures requires `aticonfig`. Catalyst, I assume, being the old name for what is now called the AMD Radeon Software suite.

I can't find any "catalyst" packages (besides the Perl framework) or "aticonfig" anywhere in the Fedora or RPM Fusion package list. And I ran across this Arch wiki page saying Catalyst was [removed from their official repositories](https://wiki.archlinux.org/index.php/AMD_Catalyst), because it had ceased development and requires an older unsupported version of Xorg.

So, is this note about requiring aticonfig only applicable for legacy software/hardware which would have used AMD Catalyst? Would modern AMD graphics still require Catalyst and `aticonfig` to display temperatures in Freon?

UshakovVasilii, 2018-11-18:
Hello, I have no AMD GPU now. Please update wiki if you can

terrycloth, 2018-11-18:
I just now edited the wiki's dependencies page some more, to make it a little cleaner and less repetitive... But hopefully someone else will see the bit about Catalyst, because I still don't know what the situation there is.

Currently, the Freon wiki says it requires the `aticonfig` command. If the Catalyst suite has been discontinued, then based on the AMD-related packages currently available in Fedora, I might guess the dependencies for modern AMD cards would be the driver (`xorg-x11-drv-amdgpu`) and then possibly [`radeontop`](https://github.com/clbr/radeontop).

I'd like an AMD GPU for my next computer, but that won't be for a while, so I can't test how well Freon works with AMD graphics, if at all. But if Catalyst is discontinued and being removed everywhere, then this issue is probably bigger than the documentation -- probably Freon itself would need to be edited to use `radeontop` or whatever instead.

BloodyIron, 2019-04-27:
Yeah this really needs to get updated. AMDGPU has been the main GPU driver for several years now, and catalyst sucks and is discontinued. I can't get AMD GPU info at all with Freon.

arigit, 2019-05-11:
Using Radeon with the amdgpu driver on ubuntu - up until 18.10, Freon was displaying the GPU temperature. With the upgrade to 19.04, it not longer does.

running sensors, I can see the temperature is there:

```
# sensors
asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

amdgpu-pci-0200
Adapter: PCI adapter
vddgfx:       +0.85 V  
fan1:             N/A  (min =    0 RPM, max = 4600 RPM)
temp1:        +40.0°C  (crit = +94.0°C, hyst = -273.1°C)
power1:       14.12 W  (cap =  48.00 W)

coretemp-isa-0000
Adapter: ISA adapter
.......

```
Anyway for Freon to use this input?
Since Freon already uses lm-sensors (I think) for a bunch of other stuff wouldn't it make sense for it to use this too (I assume there is no real dependency with aticonfig)

All - any workaround in the interim?

UshakovVasilii, 2019-05-11:
@arigit please provide `sensors -j` output

arigit, 2019-05-12:
@UshakovVasilii thanks for looking into this. The output follows below. The 'amdgpu' is prominent there. This is in a default ubuntu installation with the open source driver (I believe the old proprietary driver has been retired by now as amd went fully open source). See that the driver is also reporting power consumption and fan speed (in my case, I manually removed the fan from the card and replaced it with a passive cooling solution, that is probably why it shows an error)

It would be nice to get out of the box support for AMD graphics in Freon

```~> sensors -j
{
   "asus-isa-0000":{
      "Adapter": "ISA adapter",
      "cpu_fan":{
         "fan1_input": 0.000
      }
   },
   "amdgpu-pci-0200":{
      "Adapter": "PCI adapter",
      "vddgfx":{
         "in0_input": 0.850
      },
      "fan1":{
ERROR: Can't get value of subfeature fan1_input: Can't read
,
         "fan1_min": 0.000,
         "fan1_max": 4600.000
      },
      "temp1":{
         "temp1_input": 34.000,
         "temp1_crit": 94.000,
         "temp1_crit_hyst": -273.150
      },
      "power1":{
         "power1_average": 8.235,
         "power1_cap": 48.000
      }
   },
   "coretemp-isa-0000":{
      "Adapter": "ISA adapter",
      "Package id 0":{
         "temp1_input": 39.000,
         "temp1_max": 80.000,
         "temp1_crit": 100.000,
         "temp1_crit_alarm": 0.000
      },
      "Core 0":{
         "temp2_input": 37.000,
         "temp2_max": 80.000,
         "temp2_crit": 100.000,
         "temp2_crit_alarm": 0.000
      },
      "Core 1":{
         "temp3_input": 39.000,
         "temp3_max": 80.000,
         "temp3_crit": 100.000,
         "temp3_crit_alarm": 0.000
      },
      "Core 2":{
         "temp4_input": 39.000,
         "temp4_max": 80.000,
         "temp4_crit": 100.000,
         "temp4_crit_alarm": 0.000
      },
      "Core 3":{
         "temp5_input": 38.000,
         "temp5_max": 80.000,
         "temp5_crit": 100.000,
         "temp5_crit_alarm": 0.000
      }
   }
}
```

UshakovVasilii, 2019-05-12:
@arigit looks like it was regression https://github.com/UshakovVasilii/gnome-shell-extension-freon/commit/da3af4da27169cdde1bf3eb0dbd5c1d1b77f351f

Please verify fix

arigit, 2019-05-12:
Tested - not working. Did a little troubleshooting and just realized that Freon is not picking up any output from 'sensors -j', not only missing amdgpu but also everything else (like the CPU temperatures), see screenshot:

[image]

Same output with the current release (in extensions.gnome.org) and also with the latest master here in github including the changes. Looks like the problem is in sensorsUtil.js, not able to parse any output at all. This is in Ubuntu 19.04 (latest) with vanilla install.

I suspect the line with ERROR line in sensors -j output is breaking the parsing. I am sure it was working in the past, but not sure at what point it stopped working.

arigit, 2019-05-12:
Tested parsing the "sensors -j" output, and it's invalid:

SyntaxError: JSON.parse: expected property name or '}' at line 14 column 1 of the JSON data

Then I tried this:

sensors -j 2> /dev/null

The error message goes away,

```> sensors -j 2> /dev/null
{
   "asus-isa-0000":{
      "Adapter": "ISA adapter",
      "cpu_fan":{
         "fan1_input": 0.000
      }
   },
   "amdgpu-pci-0200":{
      "Adapter": "PCI adapter",
      "vddgfx":{
         "in0_input": 0.900
      },
      "fan1":{
,
         "fan1_min": 0.000,
         "fan1_max": 4600.000
      },
      "temp1":{
         "temp1_input": 38.000,
         "temp1_crit": 94.000,
         "temp1_crit_hyst": -273.150
      },
      "power1":{
         "power1_average": 7.188,
         "power1_cap": 48.000
      }
   },
   "coretemp-isa-0000":{
      "Adapter": "ISA adapter",
      "Package id 0":{
         "temp1_input": 42.000,
         "temp1_max": 80.000,
         "temp1_crit": 100.000,
         "temp1_crit_alarm": 0.000
      },
      "Core 0":{
         "temp2_input": 41.000,
         "temp2_max": 80.000,
         "temp2_crit": 100.000,
         "temp2_crit_alarm": 0.000
      },
      "Core 1":{
         "temp3_input": 40.000,
         "temp3_max": 80.000,
         "temp3_crit": 100.000,
         "temp3_crit_alarm": 0.000
      },
      "Core 2":{
         "temp4_input": 42.000,
         "temp4_max": 80.000,
         "temp4_crit": 100.000,
         "temp4_crit_alarm": 0.000
      },
      "Core 3":{
         "temp5_input": 38.000,
         "temp5_max": 80.000,
         "temp5_crit": 100.000,
         "temp5_crit_alarm": 0.000
      }
   }
}
```

but the parsing is still invalid. I've been looking into lm-sensors github discussions and there is a lot of noise around this scenario.  Many different situations can produce this problem.

Could a workaround like this work: before parsing the sensors -j output, can the raw output be scanned and if any line starts with 
   "ERROR: Can't" 
then just replace that line with ''' "lm_sensors_error": 0.0   '''
I tested this and it makes the output parseable in JSON. It would work-around the lm-sensors issue with producing valid JSON.

I tried to implement this myself in sensorsUtil.js with no luck so far. My javascript skills are almost zero :(

arigit, 2019-05-12:
Here's my (hopefully temporary) workaround:

in ~/.local/bin, I created a tiny script
```$ cat sensors-ar 
#!/bin/bash
sensors -j 2>/dev/null | sed s/^,/\"lm_sensors_error\":0.0,/g
```

And then in 
~/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com
I changed this line
```
let path = GLib.find_program_in_path('sensors');
```
with this
```
let path = GLib.find_program_in_path('sensors-ar');
```

this gave me back the CPU temps as well as the GPU temp

Hoping that this workaround can be merged into Freon. The discussions in the lm-sensors have been going on for a few years so no hope that they will prevent their utility from generating invalid JSON anytime soon

arigit, 2019-05-12:
Actually it seems there is a fix coming in lm-sensors:

https://github.com/lm-sensors/lm-sensors/issues/121
https://github.com/lm-sensors/lm-sensors/commit/2c8cca3d6cd60121b401734c1a24cfec7daed4fc

from a few days ago, it's in lm-sensors master. I will likely take a while to make it to distros (first a new version of lmsensors has to be released and later distros need to pick up the new release)

UshakovVasilii, 2019-05-15:
@arigit I think it fixed now

arigit, 2019-05-17:
@UshakovVasilii thanks for the try! it didn't work, and after learning a little about how to debug gnome extensions I found the reason: the ERROR message is part of the 'sensors -j' output string and is still breaking the parsing.

here is my proposed fix for your code in master:  (sorry for my ugly code)

```
...................

  _parseSensorsOutput(sensorFilter, sensorType, gpuFlag) {
        if(!this._output)
            return [];

        let data = []
        try {
            data = JSON.parse(this._output.join(''));
        } catch (e) {
          try {
            // fix for wrong lm_sensors output
            // https://github.com/UshakovVasilii/gnome-shell-extension-freon/issues/114#issuecomment-491613545
            let lineRemoved = this._output.filter(l => l.trim() !== ',').join('\n');
            //global.log(lineRemoved);
            let errorRemoved = lineRemoved.replace(/ERROR.*Can't read/, "");
            //global.log(errorRemoved);
            data = JSON.parse(errorRemoved);
          } catch (e) {
            global.log(e.toString());
            return [];
          }
        }

    .....................
```

UshakovVasilii, 2019-05-17:
I thought `ERROR.*Can't read` is outputted to error stream. Thanks!

closed 2019-05-17 by UshakovVasilii