UshakovVasilii/gnome-shell-extension-freon #116: How to allow for lower minimum values for CPU?

rajlego, 2018-11-23:
On my X1 Extreme with an 8850H, the lowest I can put my CPU at is 18%. How can I lower it further?

UshakovVasilii, 2018-11-23:
Hello, is this problem related to this gnome-shell extension? ))

rajlego, 2018-11-24:
Yep, for the extension.

UshakovVasilii, 2018-11-24:
What process take CPU?

rajlego, 2018-11-25:
If you mean which one measures it, I'm not really sure, how would I go about checking? There is no process taking up significant CPU generally but with Freon when I try to lower minimum frequency below 18% (which works out to 781 mhz) it won't let me put it any lower.

UshakovVasilii, 2018-11-26:
I need some details, currently I don't understand the following:

- how do you change minimum frequency?
- how did you understood that freon related to  minimum frequency?

jonasmalacofilho, 2020-02-21:
@rajlego

Have you by any chance enabled the reading of the GPU temperature of a NVIDIA card?