UshakovVasilii/gnome-shell-extension-freon #118: User configurable text color

hlechner, 2018-12-07:
Type: **Enhancement / Feature Request**

An user configurable color text per sensor would be awesome.

Here an example based on MSI Afterburner (windows application):
[image]

(The MSI Afterburner can be configured to display each sensor on the tray in the user configurable color)

Here to exemplify I've created a mockup (just an image edited on gimp):
[image]

UshakovVasilii, 2018-12-08:
Hello, interesting feature, but I don't understand reasons to change sensors color. I think icon before sensor is enough to identify sensor. Maybe more reasonable to change color based on temperature value ))
But I won't do it for now, you can create pull request.

hlechner, 2018-12-08:
The idea behind it was to easily recognize it when there are too much information on the panel (of other extensions for example), but of course just the icon make it possible to identify the sensors, the color idea was about how fast/easy to find it on a crowded panel.

It's ok that it will not be priority on the freon development anytime soon and thanks for the development of this great extension.