UshakovVasilii/gnome-shell-extension-freon #119: Include Nvidia Fan RPM

vivivitus, 2019-01-04:
Hi, I'm wondering if you could include the fan speed of Nvidia graphic cards in future versions. The command I'm using to read the current fan speed is `nvidia-settings -q GPUCurrentFanSpeedRPM` and because you already use _nvidia-settings_ for returning the temperature, it could perhaps be only a small code addition. Sadly I don't know any js.

Strykar, 2020-12-15:
@UshakovVasilii So I took a look at https://github.com/UshakovVasilii/gnome-shell-extension-freon/blob/master/freon%40UshakovVasilii_Github.yahoo.com/nvidiaUtil.js and it appears you are calling `nvidia-smi` which does not report the fan speed in RPM, just a percentage of max speed.

This max speed would be different for different chipsets and even models of the same chipset depending on the AIB.

```
$ nvidia-smi --help-query-gpu|grep fan
"fan.speed"
The fan speed value is the percent of the product's maximum noise tolerance fan speed that the device's fan
is currently intended to run at.
This value may exceed 100% in certain cases. Note: The reported speed is the intended fan speed.
If the fan is physically blocked and unable to spin, this output will not match the actual fan speed.
Many parts do not report fan speeds because they rely on cooling via fans in the surrounding enclosure.
```
As @vivivitus suggested, `nvidia-settings` does show it, but apparently this cannot be done via `nvidia-smi` at all outside of some ugly hacks that guesstimate fan RPM based off the percentage indicated?
Or am I missing something?

jonasmalacofilho, 2020-12-15:
The switch to `nvidia-smi` was originally intended to remove annoying spikes in CPU temperatures caused by `nvidia-settings` (see #162, or as @Strykar has already noticed, 4833a2e).  That said, these days I'm seeing these spikes with `nvidia-smi` as well, so I'm fine (as the one that proposed and implemented the previous transition) with you deciding to give `nvidia-settings` another try.

Strykar, 2020-12-15:
Someone on [r/nvidia](https://www.reddit.com/r/nvidia/comments/kdjdjz/nvidiasmi_linux_how_to_get_fan_rpm_not_speed/gfxdadh?utm_source=share&utm_medium=web2x&context=3) suggested the [NVIDIA Management Library (NVML)](https://developer.nvidia.com/nvidia-management-library-nvml) as an alternative.

"`nvmlUnitGetFanSpeedInfo` API should return fan speed in RPM via the `nvmlUnitFanInfo_t` structure.
This is a C library with official Perl and Python bindings."

Seeing CPU spikes in both `nvidia-smi` and `nvidia-settings` with repetitive polling, would NVML be a worthwhile alternative to explore?

jonasmalacofilho, 2020-12-16:
Unfortunately, that and other "unit" infos are only available for "S-class Tesla enclosures".

Strykar, 2020-12-16:
Oh? nvidia-smi uses NVML to display its info from that link, it seemed it would work.

jonasmalacofilho, 2020-12-16:
Right, but NVML and nvidia-smi differentiate between GPU and "UNIT" attributes.  Fan speeds as *intended* percentages are available for all cards, but the actual fan state and rpm values are "S-class" only infos.

```
$ man nvidia-smi
...
   -u, --unit
       Display  Unit data instead of GPU data.  Unit data is only available for NVIDIA
       S-class Tesla enclosures.
...
GPU ATTRIBUTES
...
   Fan Speed
       The fan speed value is the percent of the product's maximum noise tolerance fan
       speed that the device's fan is currently intended to run at.   This  value  may
       exceed  100%  in  certain  cases.  Note: The reported speed is the intended fan
       speed.  If the fan is physically blocked and unable to spin, this  output  will
       not  match  the  actual fan speed.  Many parts do not report fan speeds because
       they rely on cooling via fans in the surrounding enclosure.  For  all  discrete
       products with dedicated fans.
...
UNIT ATTRIBUTES
...
   Fan Info
       Fan readings for the unit.  A reading is provided for each fan, of which  there
       can be many.  For all S-class products.

       State          The state of the fan, either "NORMAL" or "FAILED".

       Speed          For a healthy fan, the fan's speed in RPM.
 ```

Strykar, 2020-12-16:
Thanks for clearing that up, so `nvidia-settings` seems to be the only way? I'm so getting an AMD card next.

jonasmalacofilho, 2020-12-16:
Well, that or patching the NVIDIA kernel driver...

I'm getting an AMD card next too.