UshakovVasilii/gnome-shell-extension-freon #122: Support and respond to `fault` attributes

jonasmalacofilho, 2019-02-27:
While working on a new hwmon driver I learned that drivers can, besides returning error codes (see: #66), set fault attributes (e.g. `temp3_fault = 4`) to specific sensors.  According to the [hwmon sysfs interface](https://github.com/torvalds/linux/blob/7d762d69145a54d169f58e56d6dac57a5508debc/Documentation/hwmon/sysfs-interface#L683-L693):

> When this boolean has value 1, the measurement for that channel should not be trusted.

Fortunately, `sensors -j` reports clearly, unlike in #66:

```
$ sensors -j 'kraken-*'
{
   "kraken-hid-3-20":{
      "Adapter": "HID adapter",
      "AIO Fan Speed":{
         "fan1_input": 834.000
      },
      "AIO Pump Speed":{
         "fan2_input": 1992.000
      },
      "AIO Liquid Temperature":{
         "temp1_input": 27.900,
         "temp1_fault": 1.000
      }
   }
}
```

I think that handling this would be a nice addition, specially for critical devices, I would like to submit a pull request for this.

However, I'm not sure what should happen in the UI when a sensor is at fault.  What do you think?

UshakovVasilii, 2019-03-07:
I think `⚠` is OK for `*_fault` sensors

UshakovVasilii, 2019-05-24:
Pull request merged

closed 2019-05-24 by UshakovVasilii