UshakovVasilii/gnome-shell-extension-freon #125: GPU temp not showing in taskbar but extension gets the data

Kallestofeles, 2019-03-14:
Greetings,

The issue I have is that the Nvidia GPU data is received by the extension, but is not shown in the titlebar despite being enabled in the options. It seems that the extent of the titlebar item is squeezed together and that is why it only shows the CPU temp and three dots. Has anyone encountered this and know a solution?

OS: Arch + GNOME Shell 3.30.2
CPU: Ryzen 2700X
GPU: 2080Ti

[image]

UshakovVasilii, 2019-03-14:
Looks like `GeForce RTX 2080 Ti` element is not checked
[image]

Kallestofeles, 2019-03-15:
My god I feel stupid. Thank you for pointing out the obvious! Got it perfectly showing up now. :)

closed 2019-03-15 by Kallestofeles