UshakovVasilii/gnome-shell-extension-freon #126: "JS ERROR" when loading in GNOME 3.32

Leeo97one, 2019-03-16:
This extension is currently unusable in GNOME 3.32.
```
mars 16 13:38:00 gnome-shell[1749]: JS ERROR: TypeError: this._construct is not a function
                                    newClass@resource:///org/gnome/gjs/modules/_legacy.js:114:32
                                    _appendMenuItems@/home/leeo/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js:476:28
                                    wrapper@resource:///org/gnome/gjs/modules/_legacy.js:82:22
                                    _updateDisplay@/home/leeo/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js:415:17
                                    wrapper@resource:///org/gnome/gjs/modules/_legacy.js:82:22
                                    _updateUI@/home/leeo/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js:253:13
                                    wrapper@resource:///org/gnome/gjs/modules/_legacy.js:82:22
                                    _init/this._updateUITimeoutId<@/home/leeo/.local/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js:88:13
```

saildata, 2019-03-16:
Looks like it's caused by https://gitlab.gnome.org/GNOME/gjs/issues/233.

First example I've seen of a fix for 3.32 over at: [gnome-shell-extension-sensory-perception](https://github.com/HarlemSquirrel/gnome-shell-extension-sensory-perception/commit/50f3b6f8b310babf2be475d4af88d43f66ba0505)

Thanks for the awesome extension! It's one that I put on every Gnome install :+1:

UshakovVasilii, 2019-03-18:
Oh, backward incompatibility again  :angry:

UshakovVasilii, 2019-03-20:
What the hell?
``` js
const PanelMenu = imports.ui.panelMenu;
const Main = imports.ui.main;

class FreonMenuButton extends PanelMenu.Button {

    constructor(){
        super();

        this.foo();
    }

    foo() {
    }

};

let freonMenu;

function enable() {
    freonMenu = new FreonMenuButton();
    Main.panel.addToStatusArea('freonMenu', freonMenu);
}

function disable() {
    freonMenu.destroy();
    freonMenu = null;
}
```
Error: 
```
Extension "freon@UshakovVasilii_Github.yahoo.com" had error: TypeError: this.foo is not a function
```

pcm720, 2019-03-20:
@UshakovVasilii 
`PanelMenu.Button` is a GObject now. So, the class must be initialized like this:
```
var FreonMenuButton = GObject.registerClass( class FreonMenuButton extends PanelMenu.Button {
     _init() {
          super._init();
     }
});
```

I don't know what GNOME Shell developers were thinking, but that was enough to break even [ES6-compliant extensions](https://github.com/JasonLG1979/gnome-shell-extensions-mpris-indicator-button/pull/3/files).

UshakovVasilii, 2019-03-20:
Done, thanks all!

daenney, 2019-03-23:
I'm on https://github.com/UshakovVasilii/gnome-shell-extension-freon/commit/98630647ddc04de8a4d419be97f86ed7773b8836 but I'm still seeing this logged:

```
JS ERROR: TypeError: this._construct is not a function
newClass@resource:///org/gnome/gjs/modules/_legacy.js:114:32
_appendMenuItems@/usr/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js:520:28
wrapper@resource:///org/gnome/gjs/modules/_legacy.js:82:22
_updateDisplay@/usr/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js:454:17
wrapper@resource:///org/gnome/gjs/modules/_legacy.js:82:22
_updateUI@/usr/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js:289:13
wrapper@resource:///org/gnome/gjs/modules/_legacy.js:82:22
_init/this._updateUITimeoutId<@/usr/share/gnome-shell/extensions/freon@UshakovVasilii_Github.yahoo.com/extension.js:121:13
```

Leeo97one, 2019-03-23:
Yep, the update is now available on the GNOME Shell Extensions website and everything is working. Good job! :raised_hands:

closed 2019-03-23 by Leeo97one