UshakovVasilii/gnome-shell-extension-freon #129: No sensors shown with new 3.32 update

brittyazel, 2019-03-22:
Hello,

I just updated Freon to the newly updated release for 3.32, and while the extension doesn't crash (which is awesome!) it also doesn't show me any of my sensors besides my HDD sensor. I have verified that my lm_sensors is set up and work properly, as this is the output of "sensors" command:

```
iwlwifi-virtual-0
Adapter: Virtual device
temp1:            N/A  

amdgpu-pci-2d00
Adapter: PCI adapter
vddgfx:       +0.75 V  
fan1:         753 RPM  (min =    0 RPM, max = 3200 RPM)
temp1:        +30.0°C  (crit = +91.0°C, hyst = -273.1°C)
power1:       32.17 W  (cap = 183.00 W)

nct6779-isa-0290
Adapter: ISA adapter
Vcore:                  +0.25 V  (min =  +0.00 V, max =  +1.74 V)
in1:                    +1.27 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
AVCC:                   +3.30 V  (min =  +2.98 V, max =  +3.63 V)
+3.3V:                  +3.30 V  (min =  +2.98 V, max =  +3.63 V)
in4:                    +1.86 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                    +1.11 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                    +1.37 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
3VSB:                   +3.46 V  (min =  +2.98 V, max =  +3.63 V)
Vbat:                   +3.28 V  (min =  +2.70 V, max =  +3.63 V)
in9:                    +0.00 V  (min =  +0.00 V, max =  +0.00 V)
in10:                   +0.87 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                   +1.08 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                   +1.67 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                   +0.94 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                   +1.82 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                   439 RPM  (min =    0 RPM)
fan2:                   600 RPM  (min =    0 RPM)
fan3:                   632 RPM  (min =    0 RPM)
fan4:                   483 RPM  (min =    0 RPM)
fan5:                   467 RPM  (min =    0 RPM)
SYSTIN:                 +29.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM  sensor = thermistor
CPUTIN:                 +30.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                 +7.0°C    sensor = thermistor
AUXTIN1:                +33.0°C    sensor = thermistor
AUXTIN2:                +21.0°C    sensor = thermistor
AUXTIN3:                -26.0°C    sensor = thermistor
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C  
PCH_CHIP_TEMP:           +0.0°C  
PCH_CPU_TEMP:            +0.0°C  
PCH_MCH_TEMP:            +0.0°C  
intrusion0:            ALARM
intrusion1:            ALARM
beep_enable:           disabled

k10temp-pci-00c3
Adapter: PCI adapter
Tdie:         +39.2°C  (high = +70.0°C)
Tctl:         +59.2°C  

```

Any idea on what is happening? Prior to 3.32 I had all the sensors and voltages showing great.

[image]

UshakovVasilii, 2019-03-22:
Can be related lm-sensors version, should be 3.5.0`sensors -j`  #101 https://github.com/UshakovVasilii/gnome-shell-extension-freon/issues/109#issuecomment-435070363

brittyazel, 2019-03-22:
I commented on the other thread. I am on lm_sensors 3.5.0. Also, an interesting note is that my Intel system shows the sensors fine, but my Ryzen system only shows HDD info

UshakovVasilii, 2019-03-23:
You can check logs `journalctl -b | grep freon` or real-time `journalctl -f`, also you can check `sensors -j` output.

I'm on Archlinux with GNOME 3.32 and extension works fine.

brittyazel, 2019-03-24:
the journalctl command didn't return anything. It was empty.

This is the return of sensors -j

```
{
   "iwlwifi-virtual-0":{
      "Adapter": "Virtual device",
      "temp1":{
ERROR: Can't get value of subfeature temp1_input: Can't read

      }
   },
   "amdgpu-pci-2d00":{
      "Adapter": "PCI adapter",
      "vddgfx":{
         "in0_input": 0.825
      },
      "fan1":{
         "fan1_input": 748.000,
         "fan1_min": 0.000,
         "fan1_max": 3200.000
      },
      "temp1":{
         "temp1_input": 27.000,
         "temp1_crit": 91.000,
         "temp1_crit_hyst": -273.150
      },
      "power1":{
         "power1_average": 33.112,
         "power1_cap": 183.000
      }
   },
   "nct6779-isa-0290":{
      "Adapter": "ISA adapter",
      "Vcore":{
         "in0_input": 0.712,
         "in0_min": 0.000,
         "in0_max": 1.744,
         "in0_alarm": 0.000,
         "in0_beep": 0.000
      },
      "in1":{
         "in1_input": 1.280,
         "in1_min": 0.000,
         "in1_max": 0.000,
         "in1_alarm": 1.000,
         "in1_beep": 0.000
      },
      "AVCC":{
         "in2_input": 3.296,
         "in2_min": 2.976,
         "in2_max": 3.632,
         "in2_alarm": 0.000,
         "in2_beep": 0.000
      },
      "+3.3V":{
         "in3_input": 3.296,
         "in3_min": 2.976,
         "in3_max": 3.632,
         "in3_alarm": 0.000,
         "in3_beep": 0.000
      },
      "in4":{
         "in4_input": 1.856,
         "in4_min": 0.000,
         "in4_max": 0.000,
         "in4_alarm": 1.000,
         "in4_beep": 0.000
      },
      "in5":{
         "in5_input": 1.112,
         "in5_min": 0.000,
         "in5_max": 0.000,
         "in5_alarm": 1.000,
         "in5_beep": 0.000
      },
      "in6":{
         "in6_input": 1.368,
         "in6_min": 0.000,
         "in6_max": 0.000,
         "in6_alarm": 1.000,
         "in6_beep": 0.000
      },
      "3VSB":{
         "in7_input": 3.456,
         "in7_min": 2.976,
         "in7_max": 3.632,
         "in7_alarm": 0.000,
         "in7_beep": 0.000
      },
      "Vbat":{
         "in8_input": 3.280,
         "in8_min": 2.704,
         "in8_max": 3.632,
         "in8_alarm": 0.000,
         "in8_beep": 0.000
      },
      "in9":{
         "in9_input": 0.000,
         "in9_min": 0.000,
         "in9_max": 0.000,
         "in9_alarm": 0.000,
         "in9_beep": 0.000
      },
      "in10":{
         "in10_input": 0.856,
         "in10_min": 0.000,
         "in10_max": 0.000,
         "in10_alarm": 1.000,
         "in10_beep": 0.000
      },
      "in11":{
         "in11_input": 1.080,
         "in11_min": 0.000,
         "in11_max": 0.000,
         "in11_alarm": 1.000,
         "in11_beep": 0.000
      },
      "in12":{
         "in12_input": 1.672,
         "in12_min": 0.000,
         "in12_max": 0.000,
         "in12_alarm": 1.000,
         "in12_beep": 0.000
      },
      "in13":{
         "in13_input": 0.936,
         "in13_min": 0.000,
         "in13_max": 0.000,
         "in13_alarm": 1.000,
         "in13_beep": 0.000
      },
      "in14":{
         "in14_input": 1.832,
         "in14_min": 0.000,
         "in14_max": 0.000,
         "in14_alarm": 1.000,
         "in14_beep": 0.000
      },
      "fan1":{
         "fan1_input": 416.000,
         "fan1_min": 0.000,
         "fan1_alarm": 0.000,
         "fan1_beep": 0.000,
         "fan1_pulses": 2.000
      },
      "fan2":{
         "fan2_input": 572.000,
         "fan2_min": 0.000,
         "fan2_alarm": 0.000,
         "fan2_beep": 0.000,
         "fan2_pulses": 2.000
      },
      "fan3":{
         "fan3_input": 599.000,
         "fan3_min": 0.000,
         "fan3_alarm": 0.000,
         "fan3_beep": 0.000,
         "fan3_pulses": 2.000
      },
      "fan4":{
         "fan4_input": 458.000,
         "fan4_min": 0.000,
         "fan4_alarm": 0.000,
         "fan4_beep": 0.000,
         "fan4_pulses": 2.000
      },
      "fan5":{
         "fan5_input": 447.000,
         "fan5_min": 0.000,
         "fan5_alarm": 0.000,
         "fan5_beep": 0.000,
         "fan5_pulses": 2.000
      },
      "SYSTIN":{
         "temp1_input": 26.000,
         "temp1_max": 0.000,
         "temp1_max_hyst": 0.000,
         "temp1_alarm": 1.000,
         "temp1_type": 4.000,
         "temp1_offset": 0.000,
         "temp1_beep": 0.000
      },
      "CPUTIN":{
         "temp2_input": 30.000,
         "temp2_max": 80.000,
         "temp2_max_hyst": 75.000,
         "temp2_alarm": 0.000,
         "temp2_type": 4.000,
         "temp2_offset": 0.000,
         "temp2_beep": 0.000
      },
      "AUXTIN0":{
         "temp3_input": 7.000,
         "temp3_type": 4.000,
         "temp3_offset": 0.000
      },
      "AUXTIN1":{
         "temp4_input": 34.000,
         "temp4_type": 4.000,
         "temp4_offset": 0.000
      },
      "AUXTIN2":{
         "temp5_input": 21.000,
         "temp5_type": 4.000,
         "temp5_offset": 0.000
      },
      "AUXTIN3":{
         "temp6_input": -26.000,
         "temp6_type": 4.000,
         "temp6_offset": 0.000
      },
      "PCH_CHIP_CPU_MAX_TEMP":{
         "temp7_input": 0.000
      },
      "PCH_CHIP_TEMP":{
         "temp8_input": 0.000
      },
      "PCH_CPU_TEMP":{
         "temp9_input": 0.000
      },
      "PCH_MCH_TEMP":{
         "temp10_input": 0.000
      },
      "intrusion0":{
         "intrusion0_alarm": 1.000,
         "intrusion0_beep": 0.000
      },
      "intrusion1":{
         "intrusion1_alarm": 1.000,
         "intrusion1_beep": 0.000
      },
      "beep_enable":{
         "beep_enable": 0.000
      }
   },
   "k10temp-pci-00c3":{
      "Adapter": "PCI adapter",
      "Tdie":{
         "temp1_input": 34.875,
         "temp1_max": 70.000
      },
      "Tctl":{
         "temp2_input": 54.875
      }
   }
}

```

dbNj4c5, 2019-03-24:
I have the same problem as brittyazel.. Hard disks show in Freon, but no CPU temperatures.

Runninng arch on Intel i7-8750h

```
$ sensors
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +45.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +45.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +42.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +42.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +42.0°C  (high = +100.0°C, crit = +100.0°C)
Core 4:        +41.0°C  (high = +100.0°C, crit = +100.0°C)
Core 5:        +41.0°C  (high = +100.0°C, crit = +100.0°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +25.0°C  (crit = +107.0°C)

iwlwifi-virtual-0
Adapter: Virtual device
temp1:            N/A  

pch_cannonlake-virtual-0
Adapter: Virtual device
temp1:        +40.0°C  

```

$ journalctl -b | grep -i freon
is empty

```
$ sensors -j
{
   "coretemp-isa-0000":{
      "Adapter": "ISA adapter",
      "Package id 0":{
         "temp1_input": 44.000,
         "temp1_max": 100.000,
         "temp1_crit": 100.000,
         "temp1_crit_alarm": 0.000
      },
      "Core 0":{
         "temp2_input": 44.000,
         "temp2_max": 100.000,
         "temp2_crit": 100.000,
         "temp2_crit_alarm": 0.000
      },
      "Core 1":{
         "temp3_input": 44.000,
         "temp3_max": 100.000,
         "temp3_crit": 100.000,
         "temp3_crit_alarm": 0.000
      },
      "Core 2":{
         "temp4_input": 42.000,
         "temp4_max": 100.000,
         "temp4_crit": 100.000,
         "temp4_crit_alarm": 0.000
      },
      "Core 3":{
         "temp5_input": 43.000,
         "temp5_max": 100.000,
         "temp5_crit": 100.000,
         "temp5_crit_alarm": 0.000
      },
      "Core 4":{
         "temp6_input": 42.000,
         "temp6_max": 100.000,
         "temp6_crit": 100.000,
         "temp6_crit_alarm": 0.000
      },
      "Core 5":{
         "temp7_input": 44.000,
         "temp7_max": 100.000,
         "temp7_crit": 100.000,
         "temp7_crit_alarm": 0.000
      }
   },
   "acpitz-acpi-0":{
      "Adapter": "ACPI interface",
      "temp1":{
         "temp1_input": 25.000,
         "temp1_crit": 107.000
      }
   },
   "iwlwifi-virtual-0":{
      "Adapter": "Virtual device",
      "temp1":{
ERROR: Can't get value of subfeature temp1_input: Can't read

      }
   },
   "pch_cannonlake-virtual-0":{
      "Adapter": "Virtual device",
      "temp1":{
         "temp1_input": 40.000
      }
   }
}

```
HTH.

UshakovVasilii, 2019-03-25:
Output has error:
> ERROR: Can't get value of subfeature temp1_input: Can't read

Need to skip broken sensor https://github.com/UshakovVasilii/gnome-shell-extension-freon/wiki/FAQ#error-in-sensors-output

dbNj4c5, 2019-03-25:
Ugh. Sorry, all working now. 

Thanks for your help - and patience!

brittyazel, 2019-03-26:
This also works for me! Thank you

closed 2019-03-26 by brittyazel

ah-, 2019-04-17:
Thanks for the workaround!
Is there a way of fixing this? Could it just print the error to stderr instead?

UshakovVasilii, 2019-04-18:
I'm not sure, I cannot reproduce it ((