UshakovVasilii/gnome-shell-extension-freon #130: AMDGPU: fan, power and voltage missing

daenney, 2019-03-31:
The amdgpu reports fan, power and voltage too but it seems Freon only catches the temperature aspect:

```
   "amdgpu-pci-8300":{
      "GPU: Voltage":{
         "in0_input": 0.756
      },
      "GPU: Fan":{
         "fan1_input": 1117.000,
         "fan1_min": 0.000,
         "fan1_max": 3500.000
      },
      "GPU: Temperature":{
         "temp1_input": 45.000,
         "temp1_crit": 91.000,
         "temp1_crit_hyst": -273.150
      },
      "GPU: Power":{
         "power1_average": 5.000,
         "power1_cap": 222.000
      }
   },
```

daenney, 2019-03-31:
I think the problem is https://github.com/UshakovVasilii/gnome-shell-extension-freon/blob/master/freon%40UshakovVasilii_Github.yahoo.com/sensorsUtil.js#L19-L21.

UshakovVasilii, 2019-03-31:
@daenney please verify fix

daenney, 2019-03-31:
Works like a charm. Thank you very much for the super quick fix!

closed 2019-03-31 by daenney